OrcaSlicer/OrcaSlicer #4354: Request for Experimental Feature: Seam/Scarf overhang threshold

Ic33s, 2024-03-06:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Scarf works great, but produces problems in overhang layers (e.g. filets at the bottom ... and yes I know bad design for 3D printer, but still often used in downloaded models). At some level it just is better to deactivate scarf completely. 

The scarf produces "holes" and rough, not-clean surface in the problem cases.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

After some testing with the dev version of orca slicer and the scarf functionality. I found that scarf values are most dependent on the overhang angle and I want to suggest to add some overhang threshold settings.
E.g. 
For normal walls until a given overhang angle I want to use scarf with length=30 and steps = 15. (Flawless seam in my testprints)
For an other overhang angle range I want to apply length=20 and steps = 10 (Default value => Not really used by me after switching to 30-15 and not useable for my overhang layers either)
For an other overhang angle range I want to apply length=5 and steps=3 (Usable as part of bottom filets to improve quality)
For an other overhang angle range I want to disable scarf completly. (Sometimes scarf just don't work)

In a first approach it would be sufficiant if the overhang angle is the max angle of the layer and scarf is applied according to the user settings.

Maybe it is also possible to apply different values automaticly and avoid regions with a specific overhang angle for the scarf and relocate the scarf (seam) to an other position of the layer.

### Describe alternatives you've considered

1. I worked with modifier objects at the moment to apply different seam/scarf settings to different layer, but it's a pain to possition and preparing each print specificly.

dabstractor, 2024-03-07:
I second this. I've been trying to print a funnel that is 2 lines thick and I can either get the countour seam to look good or the hole seam, as I adjust the values to fix one the other one gets worse. I think having some kind of scaling factor for these values that applies negatively to overhangs (>90deg) and positively on layers that are smaller than the previous layer (<90deg) to be able to keep from setting so many different independent values for various ranges. Of course I have no idea if they will scale identically and there's value in having precise control over these settings in some situations but there is significant room for improvement to this very nice feature here. As it is, it seems to work best on vertical cylinders or tuned for either a top side or a bottom side of an angled print but getting them to both look decent has so far been unachievable for me.

NeighNeighNeigh, 2024-05-01:
Some parts have areas of overhang on the same perimeter that has areas with no overhang. It would be extra nice if the scarf algorithm could slide the scarf around the perimeter to avoid the areas of overhang. This would allow such parts to still benefit from the scarf rather than completely disabling it.

closed 2024-08-07 by github-actions[bot]