OrcaSlicer/OrcaSlicer #4355: Organic Support walls not aligning for Tall objects

sunnyc90, 2024-03-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

CPU: AMD 5800x3d
RAM: 64GB
GPU: Nvidia RTX 3090

### Printer

Neptune 4 Max

### How to reproduce

Slice with any large object that need organic tree supports. Changning angle settings does not help there are still unsupported walls that causes the supports to to fail. 

### Actual results

[image]
[image]

### Expected results

Walls should align. 

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Due to the nature of this project, I am unable to share the actual project files however one should be able to reproduce the issue by printing large tall objects that need supports.

skl111, 2024-03-12:
Its probably related to this bug as well https://github.com/SoftFever/OrcaSlicer/issues/4158

tintad, 2024-03-18:
I have the same issue here. I tried all the settings that I could but nothing fixed this

[image]

fritzw, 2024-06-20:
Not stale. This is still present in 2.1.0-RC.

Here's a simple 3mf file that reproduces the issue. Apparently, the algorithm can't handle objects that intrude on the planned path of the trunk: [tree_support_misalignment.3mf.zip](https://github.com/user-attachments/files/15920265/tree_support_misalignment.3mf.zip)

[image]

skl111, 2024-06-20:
Per testing by softfever

https://github.com/SoftFever/OrcaSlicer/issues/4158#issuecomment-2135580016

This is a bug upstream in Prusa\Bambu slicers as well.

fritzw, 2024-09-19:
The issue is still relevant, not stale.

fritzw, 2024-12-19:
nope

tlhintoq, 2024-12-29:
Still happening in 2.3.0-dev.  Upstream or not this needs to be addressed.  Fixing it would help make Orca stand out when the upstream source couldn't or wouldn't fix it.

[image]

tlhintoq, 2025-03-30:
Please for the love of us people that would like to actually use these supports - Fix them so they are printable.

[image]

fritzw, 2025-06-29:
No, it's not stale.

tlhintoq, 2026-10-30:
{bot nudge}