OrcaSlicer/OrcaSlicer #4370: Better fan control for high temp materials like ASA

HakunMatat4, 2024-03-08:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have been printing ASA/ASA-CF a lot and as you might know, you should never use 100% cooling or you will be printing foam, you still wanna have a decent bridging.
Still, OrcaSlicer lacks that fan management, so for example, this is all fan related settings I can see.

[image]

Yet, there is no way to set bridge to 100% cooling, 33% cooling will look like crappy haha
I mean, it will slow down but the cooling speed isn't optimized.
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

PrusaSlicer on the other hand does allow you to set bridge ONLY to 100% even so I didn't manage to get that to work, that is not the point of this ticket.
It would be awesome if OrcaSlicer has this extra fan setting and working hahaha

[image]

sleewok, 2024-04-11:
Can we please get independent fan speed for bridges implemented? I am printing high speed TPU and Orcaslicer is a POS for this. Setting a high fan speed for bridges pushes the filament all the way down. Bridge fan speed should not be related to overhang fan speed. Overhangs have an underlying layer of filament to rest against. Bridges do not. They are COMPLETELY unrelated.

kayboy-bebop, 2024-04-27:
Second the above points. Orca has very detailed settings for overhang speed, but fan speed can't be controlled at the same level as Prusa Slicer. I find myself missing this feature quite a lot, especially with high speed profiles.

sleewok, 2024-05-23:
Adding on to this. We need independent fan speeds for internal and external bridges. as well as overhangs.

kayboy-bebop, 2024-05-23:
Figure I might as well add that support interface fan speed would be fantastic to weaken the bond between the part and the support without an excessive gap. I think SuperSlicer is the best reference to aim for with all this. Pretty much any extrusion type can have it's fan speed overridden and the fan speed behaviour is summarised in text form below. Only thing it misses is a more fine tuned overhang setting.
[image]

sleewok, 2024-05-23:
> Figure I might as well add that support interface fan speed would be fantastic to weaken the bond between the part and the support without an excessive gap. I think SuperSlicer is the best reference to aim for with all this. Pretty much any extrusion type can have it's fan speed overridden and the fan speed behaviour is summarised in text form below. Only thing it misses is a more fine tuned overhang setting that Prusa has. ![superslicer_fanspeed_example](https://private-user-images.githubusercontent.com/115389530/333066307-948f769f-7360-4f97-ac04-95d48c085030.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY0NDY1NjAsIm5iZiI6MTcxNjQ0NjI2MCwicGF0aCI6Ii8xMTUzODk1MzAvMzMzMDY2MzA3LTk0OGY3NjlmLTczNjAtNGY5Ny1hYzA0LTk1ZDQ4YzA4NTAzMC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUyM1QwNjM3NDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iMmNlNjVmNjA0MmY4MzZiNzAxZjIxMDI3MGI1MzQwNWU5MWE3YmY3OTUwZWE1ZGNhMWYzMmFmZTRhNjZjYzc3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.xhNnnTOrfASwC2LChY8YV0XLfeaeDu-ApVhoUmbMTJo)

This feature has already been implemented.

[image]

kayboy-bebop, 2024-05-27:
> > Figure I might as well add that support interface fan speed would be fantastic to weaken the bond between the part and the support without an excessive gap. I think SuperSlicer is the best reference to aim for with all this. Pretty much any extrusion type can have it's fan speed overridden and the fan speed behaviour is summarised in text form below. Only thing it misses is a more fine tuned overhang setting that Prusa has. ![superslicer_fanspeed_example](https://private-user-images.githubusercontent.com/115389530/333066307-948f769f-7360-4f97-ac04-95d48c085030.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY0NDY1NjAsIm5iZiI6MTcxNjQ0NjI2MCwicGF0aCI6Ii8xMTUzODk1MzAvMzMzMDY2MzA3LTk0OGY3NjlmLTczNjAtNGY5Ny1hYzA0LTk1ZDQ4YzA4NTAzMC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUyM1QwNjM3NDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iMmNlNjVmNjA0MmY4MzZiNzAxZjIxMDI3MGI1MzQwNWU5MWE3YmY3OTUwZWE1ZGNhMWYzMmFmZTRhNjZjYzc3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.xhNnnTOrfASwC2LChY8YV0XLfeaeDu-ApVhoUmbMTJo)
> 
> This feature (fan speed for support interface) has already been implemented.
> 
> ![image](https://private-user-images.githubusercontent.com/304940/333256920-b81e75ee-3ac8-4e53-bd7a-99fcb474379f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY3Nzg2MTcsIm5iZiI6MTcxNjc3ODMxNywicGF0aCI6Ii8zMDQ5NDAvMzMzMjU2OTIwLWI4MWU3NWVlLTNhYzgtNGU1My1iZDdhLTk5ZmNiNDc0Mzc5Zi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUyN1QwMjUxNTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05NGNhNTEzMjZjNmM4MTkyMzhkYzZkMWQ2ZTA5ZGVlZTU5OTU4NGNiNjEzYWFiMmYyNDRmYTI5MjIyMzg4NmQzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ROV4csX2imu_buI9a9pZZsk5oG6U4geNKhHNwSYgAk0)

Apologies. Don't know how I missed that.

closed 2024-09-02 by github-actions[bot]