OrcaSlicer/OrcaSlicer #4371: Can't create new nozzle for existing printer

evilhamsterman, 2024-03-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

Windows

### OS Version

Win 11

### Printer

AnkerMake M5C

### How to reproduce

1. Select Create Printer from printer drop down
2. Check the box Create Nozzle for Existing Printer
3. Try to select existing Printer

### Actual results

There is no drop down in the existing printer list

### Expected results

My existing printer would be there

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14532352/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Jorzuiz, 2024-03-20:
Bug still persists on OrcaSlicer 2.0.0

Windows 10 19045
Intel-i76700k
Nvidia GTX1070

ChopperOriginals, 2024-03-29:
I am having the same issue trying add nozzles to the KE

SuhbatDaRobot, 2024-04-08:
Came here to report the same, version 1.9.1

foreachthing, 2024-04-08:
I've never used this function before.
Please tell me, how is that (or would be) different from "Edit preset", change Nozzle-ø in Extruder and save it with a new name?

SuhbatDaRobot, 2024-04-08:
There is no difference, the workflow is different, thank you for the workaround solution foreachthing.

evilhamsterman, 2024-05-02:
@foreachthing as was said there's no difference in functionality, but the option as it stands is broken. Either it should be fixed or removed

closed 2024-08-09 by github-actions[bot]

atrt7, 2024-08-20:
I still have this issue in the latest 2.1.1, I also can't use the create printer function as when I select a vendor it says "The model was not found, please reselect the vendor"

Bassadin, 2024-11-14:
Same here, also getting "The model was not found; please reselect the vendor"

UMineiro, 2024-11-23:
Same thing as with art7 above in latest 2.2.0 on Arm macOS Sequoia 15.1

navi21, 2024-11-24:
issue still in 2.2.0!!

B-Man1, 2024-12-18:
I also have this issue

boinappi, 2025-01-03:
Just had this late night "hey give this slicer a shot"-moment. First thing happening: can't create nozzle for printer because of no printer selectable.

Windows
2.3.0-dev
Build f0e773f