OrcaSlicer/OrcaSlicer #4379: PA Pattern calibration

jumika, 2024-03-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

Windows

### OS Version

11

### Printer

Ender 3 S1 Pro

### How to reproduce

Generate PA calibration pattern.

### Actual results

Go to process / global / speed: Acceleration are set to 500. Outer wall speed set to 96.
Go to process / objects / Cube object: Acceleration values are set to 500.

### Expected results

Acceleration and speed  values should not change.

### Project file & Debug log uploads

[S1P-PA-pattern-bug-projects.zip](https://github.com/SoftFever/OrcaSlicer/files/14538150/S1P-PA-pattern-bug-projects.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

I'm using a modified profile, but this also affects the default S1 pro profile, and i even tried with a QIDI xmax 3 profile, and that also have this issue.
But this does not happen on my Sovol SV06 profile.

igiannakas, 2024-03-08:
With the latest version it sets the acceleration and print speed of your external perimeter, as this is what you optimise PA against. If they are not set (0) it uses the internal perimeter. Else it uses the default speed and accel. Finally if your print speeds are slow (under 100mm/sec) it sets it to 100mm/sec as the PA test will yield wrong results with low speeds.

If you want to test higher speeds and accels, you can set them ALL to the same value manually.

jumika, 2024-03-08:
That makes sense, and it does work as you described. However i still think it is not working perfectly because my acceleration settings are turned off when i generated the calibration pattern.
It looks like this:
[image]
Yes, it says 500 in the disabled outer wall field, but using these exact settings does not output any acceleration gcode on slicing.

Is this behaviour documented somewhere? I think it would be a good idea to show this as a help text in the generate pa calibration window.

igiannakas, 2024-03-08:
When I test PA pattern with 0 acceleration under normal printing setting it takes the outer wall accel for me and uses it. You should be able to see a set_velocity_limit ACCEL command in the gcode. Yes its not as slick - it should probably not set any accelerations at all in that scenario I think.

jumika, 2024-03-08:
Actually i get `SET_VELOCITY_LIMIT ACCEL=500 ACCEL_TO_DECEL=250` in my gcode even if outer wall accel is set to 1500 before i run the calibration tool.
The settings in the ui gets updated like this:
[image]
[image]
[Cube_PETG_8m48s.gcode.txt](https://github.com/SoftFever/OrcaSlicer/files/14542541/Cube_PETG_8m48s.gcode.txt)

igiannakas, 2024-03-09:
You need to save the profile - else it won’t take the changed values into account. It reads from the saved values not the ones on the screen when creating the test. Unfortunately this is a limitation of the way the values are stored and handled.

jumika, 2024-03-09:
Unfortunately it does not make a difference. The exact steps i've done:

1. New project.
2. Select printer, material, base profile.
3. Set all acceleration values to 1500 in profile.
4. Save profile as user preset.
5. Generate PA pattern with default settings. (I have the proper accel value 1500 everywhere in Process / Object)
6. Generate GCODE.

I still get this in the output: `SET_VELOCITY_LIMIT ACCEL=500 ACCEL_TO_DECEL=250`

Should i provide the project files again with this exact method?

igiannakas, 2024-03-09:
What value do you have under your printer settings? They can act as a limiter to what orca is allowed to emit from the profiles.

jumika, 2024-03-10:
That's it. I had maximum acceleration for extruding set to 500. Which does not have an effect on ordinary slicing because i'm on klipper gcode flavor, but made a difference in this case. Thank you very much for your help.
Anyway i still feel using this feature is cumbersome because of the confusing settings requirements.
The calibration gcode generation takes the outer wall settings as a base even if it is disabled (by setting normal accel value to 0 in the processing profile) and limits the acceleration by the printer profile / motion ability even if the checkbox emit limit to gcode says these settings are ignored when using klipper flavor. Which it does in ordinary slicing, but not in the PA pattern case.
At least we have this conversation here for future reference. Thanks for your help again!

igiannakas, 2024-03-10:
The printer limits are not ignored in ordinary slicing - it was in your specific case as you had acceleration set to 0 ;). I’ll work on improving this test, hopefully for the subsequent release.

closed 2024-06-17 by github-actions[bot]

ShantanuNair, 2025-05-22:
This is a tad unclear to me @igiannakas - would you mind clarifying? The printer limits ignored when using klipper firmware is what the tooltip says. And in normal printing, speeds/accels are used as set by the user, or, if they set it to 0, acceleration is left to klipper. In regular printing, if accel is set to 0, then no impact from printer limits settings, but if you are running the PA test, then, surprise, you have to look at the motion ability settings where extruding accel is likely limiting the test. Am I understanding correctly?

igiannakas, 2025-05-22:
Your PA test must be set up to the max recommended by IS which is by default below the max machine limits in both the printer config and slicer printer settings.

Doing a Pa test above the IS max means what you’re seeing as Pa issues are actually IS smoothing artefacts.

ShantanuNair, 2025-05-22:
I haven't yet used IS, though it's up next. The reason I ask is that, when I started printing, I put in 5k max acceleration in the klipper config, and then 0 for orca's acceleration profile settings, which caused the rest to be greyed out. Then, I noticed that there were no Set_velocity macros in the gcode for my PA Test(!). I noticed this as my mainsail UI was always stuck at 500mm2/s accel after a PA test, if I hadn't remembered to set it back after the test, slowing down my next print. Then I looked into the motion ability pane, which I thought I could ignore, since I was under the impression those settings wouldn't matter since I'm on klipper, and that acceleration wouldn't be set there. But max accel while extruding was 500. On switching that to 5000, I started seeing the SET_VELOCITY custom gcode in the preview. Shall I run a few tests then share a project in an issue if it seems like it is worth looking into?

igiannakas, 2025-05-23:
No don’t do that…

Run Is. It will tell you the max accel with limited smoothing. This doesn’t mean that number should be used everywhere.

For outer walls use half of that. For inner walls 2/3rds or the same as outer walls (if shooting for max quality).  For solid infill use the IS result. For sparse infill you can use double the IS result as smoothing there doesn’t matter. For travel use the max accel that your printer is capable off (derate by 20-30% for safety). Bridge accel should be equal to outer wall accel to avoid bulging after and before the bridge. 

That way you get better quality prints as you’re well below the smoothing target of IS. And faster ones too because you’re not limiting yourself to non smoothed accels where they don’t matter.

In your printer.cfg you should enter the value the printer is capable off mechanically. Not the IS value. I have mine at 600mm/sec and 20k accel. Walls I print at 2k, sparse infill at 10k and travel at 15k. This is on a Voron 350 with max Y accel from IS at 5k.

igiannakas, 2025-05-23:
what you need to know is that IS recommended accel is an arbitrary number that keeps corner smoothing below a certain threshold (deviation) in mm. That is the smoothing value you see in the Is charts. Going below that reduces smoothing giving you better and sharper corners. Going above that smooths out corners a bit more but that doesn’t matter for travel and sparse infill in the slightest!