OrcaSlicer/OrcaSlicer #4381: Blank dialog when starting AppImage on Linux

Area5142, 2024-03-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.0 + 1.9.1

### Operating System (OS)

Linux

### OS Version

Linux Mint 21.2 x64

### Printer

Not configured yet

### How to reproduce

1) On newly installed Linux Mint 21.2 system, start AppImage and accept SSL. Aplication terminates after this.
2) Start AppImage again and a blank setup Wizard appears:
[image]

### Actual results

A blank wizard appears and no buttons or error messages.

### Expected results

Wizard with printer setup, like in PrusaSlicer.

### Project file & Debug log uploads

In Terminal. I get this output:
```
./OrcaSlicer_Linux_V1.9.0.AppImage 

(AppImageLauncher:2368): GLib-CRITICAL **: 13:58:23.426: g_file_test: assertion 'filename != NULL' failed
[2024-03-08 13:58:30.052146] [0x00007f2a24835c80] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
loading network module at /home/user/.config/OrcaSlicer/plugins/libbambu_networking.so
error, dlerror is /home/user/.config/OrcaSlicer/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)

(orca-slicer:2389): GLib-GObject-WARNING **: 13:58:30.467: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:2389): Gtk-CRITICAL **: 13:58:30.467: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:2389): GLib-GObject-WARNING **: 13:58:30.467: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:2389): Gtk-CRITICAL **: 13:58:30.467: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:2389): GLib-GObject-WARNING **: 13:58:30.468: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:2389): Gtk-CRITICAL **: 13:58:30.468: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:2389): GLib-GObject-WARNING **: 13:58:30.469: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:2389): Gtk-CRITICAL **: 13:58:30.469: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:2389): GLib-GObject-WARNING **: 13:58:30.471: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:2389): Gtk-CRITICAL **: 13:58:30.471: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

[... 228 lines cut ...]
(orca-slicer:2389): GLib-GObject-WARNING **: 13:58:33.604: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:2389): Gtk-CRITICAL **: 13:58:33.604: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:2389): GLib-GObject-WARNING **: 13:58:33.606: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:2389): Gtk-CRITICAL **: 13:58:33.606: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:2389): GLib-GObject-WARNING **: 13:58:33.608: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:2389): Gtk-CRITICAL **: 13:58:33.608: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:2389): GLib-GObject-WARNING **: 13:58:33.608: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:2389): Gtk-CRITICAL **: 13:58:33.608: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

** (orca-slicer:2389): CRITICAL **: 13:58:33.616: Cannot register URI scheme wxfs more than once

** (orca-slicer:2389): CRITICAL **: 13:58:33.616: Cannot register URI scheme memory more than once
```

### Checklist of files to include

- [X] Log file
- [X] Project file

pellcorp, 2024-03-14:
Can work around this by adding WEBKIT_DISABLE_DMABUF_RENDERER=1 to the start of the command to run the AppImage.

novaplusplus, 2024-03-15:
Yep, that worked. Other UI stuff is still funky but I'll look around for some other extant issues on that

Area5142, 2024-03-15:
I can confirm that setting WEBKIT_DISABLE_DMABUF_RENDERER=1 works (batch file):
```
#!/bin/bash
export WEBKIT_DISABLE_DMABUF_RENDERER=1
chmod +x /home/user/tmp/OrcaSlicer_Linux_V1.9.1.AppImage
/home/user/tmp/OrcaSlicer_Linux_V1.9.1.AppImage
```
The dialogs is now visible and the startup wizard can be completed.

Area5142, 2024-03-30:
I just tried the latest version 2.0.0 RC, OrcaSlicer_Linux_V2.0.0-rc.AppImage, and that crashes after accepting SSL.
Following attempts to run the AppImage will result in a crash with no dialogs.
Specifying "export WEBKIT_DISABLE_DMABUF_RENDERER=1" makes no difference.

Tested on clean Linux Mint 21.2 x64 (same as before).

dwaq, 2024-04-02:
I'm having the same issue. I just installed orca slicer v2.0.0 for the first time on windows 10 22H2 and it boots into a blank Setup Wizard just like the first post. I also tried starting as admin with the same behavior. I don't see logs or how to start the program differently on Windows, so I can't provide much more details on what is happening.

nick-laird, 2024-04-02:
This thread talks about the issue, and a fix is to set the environment variable like they mention. It's the webkit2gtk library causing the trouble with NVIDIA drivers. So it's technically not Orca Slicer's bug but a library it uses. 
https://bugs.webkit.org/show_bug.cgi?id=261874

The workaround that has been rock solid for me is setting WEBKIT_DISABLE_DMABUF_RENDERER=1 as an environment variable before launch.

bitbooster, 2024-05-09:
I had a similar issue with OrcaSlicer 2.0.0 on Linux Mint 21.3 /fresh install, not used OS before).
Upon launching, it was getting stuck at first screen, all menus and buttons were unresponsive and even to close the application I had to kill it from terminal.
My workaround (as I read on a reddit threa it was reported working) was to download and run first OrcaSlicer 1.7.0 AppImage.
After closing v1.7.0, v2.0.0 was able to start properly.

closed 2024-08-16 by github-actions[bot]