OrcaSlicer/OrcaSlicer #4386: FR:   Allow adjustment of Seam Staggering distacne

psiberfunk, 2024-03-08:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Yes.  During my exhaustive quest to optimize scarf seams, I realized that staggered inner seams can help, but not as much as i'd like them to be able to.  The root cause of this is that using inner seams that are aligned, or closely aligned, with the start of the scarf, causes deformations on the outer walls of the scarf.   

This "landing mark" smoosh makes this particularly problematic when using the otherwise useful "inner scarfs" option that helps improve quality.  It looks like this on the first inside wall, and you can clearly see how this affects the outer wall.  As the outer wall is MOST vulnerable to issues near the scarf overlap zone (typically 20 mm long by default), and this is where this mark is landing, this is very likely causing issues with the overhang performance:

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would like to be able to adjust the stagger distance for the inner seams.. particularly so that the nearest inner wall (the one closest to the outer wall) can be offset to either before or after the "scarf zone" (20 mm by default).   

I believe that the code lives in SeamPlacer.cpp  [around here somewhere](https://github.com/SoftFever/OrcaSlicer/blob/a4bf3dabb4871a7a27dcf77d3ea4409906345705/src/libslic3r/GCode/SeamPlacer.cpp#L1579)

### Describe alternatives you've considered

I have tried almost everything under the sun to optimize this away.   I have run hundreds of experiments to optimize the seams in the challenging overhang scenario, and the best settings are [currently listed here](https://www.printables.com/model/783313/).  Howev

### Additional context

As [described here already](https://github.com/prusa3d/PrusaSlicer/issues/7872), poorly staggered inner-seams that are closely aligned have issues with even traditional walls.  Scarf seams, because of their delicate nature at low extrusions , are particularly sensitive to inner wall effects on outer walls where the scarf is , even when using inner-outer-inner wall order to try and minimize this.

[image]

Ideally we would be able to stagger the seams "away from the outer scarf" by using an adjustable offset for the start of the staggering (e.g. the nearest inner wall would be offset by X mm).

Hammerfest, 2024-03-14:
Thanks for the breakdown, i thought I was doing something wrong

CCS86, 2024-03-17:
I agree, this would be a great feature.

closed 2024-07-05 by github-actions[bot]

juricabi, 2024-08-23:
Would also like this feature, it can enable wipe on outer surface when using inner-outer-inner and outer-inner printing order, it would force longer travel and with it, wipe.
That wipe would improve normal seam in this situation.
[image]

Kellegram, 2025-07-29:
> Orca bot: this issue is stale because it has been open for 90 days with no activity.

Please stop using stale bots in your repos. They are harmful to projects. Or at least make them not do anything of contributors/owners have not responded. What is the point of this when a response from project maintainers is required to proceed???