OrcaSlicer/OrcaSlicer #4396: 1.9.1 AppImage crashes after system update

avxmw, 2024-03-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

Linux

### OS Version

openSUSE Tumbleweed

### Printer

Creality K1

### How to reproduce

1. Open AppImage via StartMenu
2. wonder why it doesn't come up
3. launch from CLI

### Actual results

Crashes with

```
(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.668: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.668: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): GLib-GObject-CRITICAL **: 10:33:18.668: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.668: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.669: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.669: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): GLib-GObject-CRITICAL **: 10:33:18.669: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.669: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:26523): GLib-GObject-CRITICAL **: 10:33:18.671: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.671: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:26523): GLib-GObject-CRITICAL **: 10:33:18.672: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.672: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:26523): GLib-GObject-CRITICAL **: 10:33:18.674: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.674: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:26523): GLib-GObject-CRITICAL **: 10:33:18.675: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.675: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:26523): GLib-GObject-CRITICAL **: 10:33:18.676: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.676: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:26523): GLib-GObject-CRITICAL **: 10:33:18.677: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.677: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.679: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.679: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): GLib-GObject-CRITICAL **: 10:33:18.679: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.679: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.680: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.680: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): GLib-GObject-CRITICAL **: 10:33:18.680: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.680: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.871: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.876: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.876: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.877: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.877: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.877: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.877: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.877: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.877: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.892: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.893: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:18.893: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:19.058: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:19.058: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:19.070: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:19.070: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:19.075: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:19.075: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:19.084: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:19.084: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:19.091: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:19.091: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:19.100: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:19.100: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:19.109: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:19.109: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:19.116: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): Gtk-CRITICAL **: 10:33:19.116: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:26523): GStreamer-WARNING **: 10:33:19.174: External plugin loader failed. This most likely means that the plugin loader helper binary was not found or could not be run. You might need to set the GST_PLUGIN_SCANNER environment variable if your setup is unusual. This should normally not be required though.

(orca-slicer:26523): libsoup-ERROR **: 10:33:19.297: libsoup2 symbols detected. Using libsoup2 and libsoup3 in the same process is not supported.

(WebKitWebProcess:26585): GStreamer-WARNING **: 10:33:21.406: External plugin loader failed. This most likely means that the plugin loader helper binary was not found or could not be run. You might need to set the GST_PLUGIN_SCANNER environment variable if your setup is unusual. This should normally not be required though.

(WebKitWebProcess:26585): libsoup-ERROR **: 10:33:21.554: libsoup2 symbols detected. Using libsoup2 and libsoup3 in the same process is not supported.
```

### Expected results

Run as usual

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

There are other issues which might be related (#4393, #4377, #4308).

AppImage should not rely on system libraries and bring all it needs was it's main feature, no?

bojangles86, 2024-03-09:
was getting this as well on fedora. seems to fixed on the nightly release

TomPcz, 2024-03-09:
I was getting the same libsoup error on Tumbleweed after a recent system update. After rebooting the machine, however, the error disappeared, and OrcaSlicer worked as expected.

avxmw, 2024-03-10:
> I was getting the same libsoup error on Tumbleweed after a recent system update. After rebooting the machine, however, the error disappeared, and OrcaSlicer worked as expected.

Thank you, that indeed worked. Feels dirty though :D

closed 2024-03-10 by avxmw

TomPcz, 2024-03-10:
Well, after a system update, you often need to restart the processes which use the updated libraries (run `zypper ps` to see the list of them) to make them run correctly. A full reboot was probably not needed, but is a quick and dirty way to restart them all :)