OrcaSlicer/OrcaSlicer #4397: Add ASA and TPU profiles

whyb33, 2024-03-09:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Can't use Orca Slicer because it will not allow me to add TPU and ASA filament profiles for my Elegoo Neptune 4. They are present and work great in the default Cura profile

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add default/preset ASA and TPU filaments as an option on all printers.

### Describe alternatives you've considered

Use Cura instead?

closed 2024-06-16 by github-actions[bot]

rojoricardo, 2024-07-12:
I guess there is no solution to this?

peperonikiller, 2024-11-13:
[https://i.imgur.com/Nk47UMW.png](url) Can confirm, there is no ASA option under material type. :(