OrcaSlicer/OrcaSlicer #4399: Is there a way to set specific settings on a project-by-project basis?

jrhager84, 2024-03-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I can't seem to find documentation anywhere that indicates I can change certain things (layer height, etc) for specific projects.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

The ability to save the settings on a per project basis for ease of opening different projects to retain certain settings.

### Additional context

Please delete if done or point me in the right direction. I may just not know what I'm looking for. Thanks!

avxmw, 2024-03-10:
File -> Save Project -> NAME.3mf saves the settings you want and restores them. Is this not what you want/need?

jrhager84, 2024-03-12:
When you try to create or open a project it asks whether to transfer the settings or revert, but it doesn't seem to support having different settings per project (where they retain and don't have to be 'transferred').

closed 2024-06-18 by github-actions[bot]