OrcaSlicer/OrcaSlicer #4406: Custom filaments vanish when decreasing nozzle size from .4 to .2

WimVDK, 2024-03-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

Windows

### OS Version

11

### Printer

Bambulab X1C

### How to reproduce

1. Create custom filament with nozzle type set to .4
2. Switch to .2 nozzle

### Actual results

Custom filament disappears

### Expected results

Custom filament should remain selectable. Just because a nozzle size changes, doesn't mean the filament suddenly is incompatible with the nozzle

### Project file & Debug log uploads

no logs for this one

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

At .4 nozzle some of my custom filaments
[image]

at .2 nozzle, all of my custom ones disappear

[image]

jsladerman, 2024-03-10:
Filament profiles are made specific to the Printer/Nozzle combo. Did you include 0.2 as an option (example below) when making the filament's profile? If not, you can also add a new nozzle option to the existing one you made (second pic)

[image]

[image]

Hope this helps!

WimVDK, 2024-03-10:
I made the first custom based off of a modified generic. Then modifed names and colors when needed. I never saw either of those screenshots.

jsladerman, 2024-03-10:
Ah makes sense. Yeah the system's a little fragmented. Best way to do it is to make a custom filament (you can still base it off an existing preset to preserve your settings) for the filaments/profiles you like to use and then it'll be easy to transfer between nozzles and printers

WimVDK, 2024-03-10:
Tried that, won't allow me to set my .4 profile for .2. I can select it for .4, .6 and .8, but i can only pick "bambu ...." for some. I just wanna easily swap them to .2 compatible. There has to be an easier way than to redo 2 dozen profiles

HAndreasH, 2024-03-11:
I strggled with the same thing as I started using Orca.
Coming from Cura it is very strange how the combination of settings for machine-filament-process works. But it seems to make sense.
For me it was not relevant to have different filament settings for 0.4 and 0.6 nozzles, so I ended up with editing the .json files for my filaments and setting "compatible_printers": []

WimVDK, 2024-03-11:
Can you post a screenshot of what and where you edited that

HAndreasH, 2024-03-12:
The way [jsladerman](https://github.com/jsladerman) described should actually work fine 
Please note that changing the files manually is actually not the best way to go!

for example a new created custom filament for a specific printer with 0.6 nozzle looks like this:

[image]

if you have more printers selected ( 0.4 and 0.6 nozzle)

[image]

this is how I did it, so the filament is visible to all printers:

[image]

closed 2024-06-18 by github-actions[bot]

WimVDK, 2024-06-18:
Reactivating this