OrcaSlicer/OrcaSlicer #4409: Creality K1C printer profile can't be used

Snoogle01, 2024-03-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0.-dec

### Operating System (OS)

macOS

### OS Version

Sonoma 14.4

### Additional system information

Apple MacPook Pro M2 pro, 14 inch, 10 Core CPU, 16 GB RAM

### Printer

Creality K1C

### How to reproduce

1. Go to printer selection
2. Select at Creality the "K1C"
3. Choose Nozzle
4. Confirm choice
5. Select filament profiles
6. Confirm

### Actual results

Although the printer is checked and active it isn't shown at the "prepare" screen and can't be chosen. Works perfectly with the normal K1 but not with the K1C

### Expected results

The K1C should be viewable and choosable on the prepare screen so that it can be used for printing.

### Project file & Debug log uploads

[Lampe_Wirk_V15_Windschild_gelb_issue.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14555164/Lampe_Wirk_V15_Windschild_gelb_issue.3mf.zip)
[Archiv.zip](https://github.com/SoftFever/OrcaSlicer/files/14555175/Archiv.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[image]
[image]
[image]

Snoogle01, 2024-03-11:
Update: It works when you complete delete the version before and also manually delete all data from the old version.

vgdh, 2024-03-12:
> Update: It works when you complete delete the version before and also manually delete all data from the old version.

can be closed?

Snoogle01, 2024-03-12:
For me yes, since it is about the pre release version and was able to deal with it. But for an official release I would expect that I could install an update without any issues. So this would be required for a final version.

closed 2024-06-18 by github-actions[bot]