OrcaSlicer/OrcaSlicer #4411: OrcaSlicer window with double titlebar

faith-healer, 2024-03-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.x and Nightly Builds

### Operating System (OS)

Linux

### OS Version

ArchLinux with KDE

### Printer

TwoTrees Sapphire SP-5

### How to reproduce

just run AppImage or AUR installation

### Actual results

[image]
double titlebar

### Expected results

[image]
single titlebar

### Project file & Debug log uploads

does not need

### Checklist of files to include

- [ ] Log file
- [ ] Project file

theblu3j, 2024-05-13:
Also have this issue. Just mildly annoying. Official AppImage and the AUR (both v2.0) both have this issue. Seems to be specific to KDE from what I can see.

fuzelet, 2024-05-14:
I also have this issue, however I am using GNOME.

[image]

## Software Information:
- **Firmware Version:**                            3501
- **OS Name:**                                     Arch Linux
- **OS Build:**                                    rolling
- **OS Type:**                                     64-bit
- **GNOME Version:**                               46
- **Windowing System:**                            X11
- **Kernel Version:**                              Linux 6.8.9-arch1-2

Luke56784, 2024-06-04:
Same issue on Ubuntu 24.04 and GNOME. Orcaslicer 2.0

Garbg, 2024-06-13:
Same issue using orcaslicer 2.0 on opensuse tumbleweed using GNOME 46 on wayland

madushan1000, 2024-09-01:
I get this too on ubuntu 24.04 gnome. It's so weird because gnome doesn't even support server side decorations. So it's the application drawing two title bars.

JaneSmith, 2024-11-11:
Same issue here on GNOME / Fedora.

Manu-pardo, 2024-11-24:
Also having the same issue in Fedora Linux 41 with the latest orca slicer stable release

Glutamat42, 2025-02-19:
You can go to Settings -> Window rules -> add new one
Give it a description, enter "OrcaSlicer" for Window class (leave it at exact match). Then add a new property "Without titel bar and border" and set it to yes. Save and start Orca slicer again. The 2nd title bar should be gone now.

I noticed it is not possible to move the window outside of the screen on left and bottom borders when dragging it through the OrcaSlicer title bar. KDE drag shortcut (depending on system win/alt + leftclick) still works. Also this change caused minor graphical artifacts outside the application when dragging the window.

[image]

closed 2025-05-29 by github-actions[bot]

MathisP75, 2025-06-30:
This is still an issue

FlorianSchwarzl, 2025-12-17:
> maybe design your apps like a normal human being and dont add unnecessary "features" like some weird window decoration.

No need for insults, also, it was inherited from BambuStudio! But I do agree that client side decorations shouldn't be a thing in the first place.