OrcaSlicer/OrcaSlicer #4419: Orca slicer doesn't set acc in gcode

Normal1no, 2024-03-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

CPU: Ryzen 5 2600X , Memory: 16gb ddr4 2600hz , GPU: Nvidea rtx 3050

### Printer

Ender 3 S1 

### How to reproduce

set accel

???

profit!
[image]

### Actual results

nothing 

### Expected results

see the acc changes 

### Project file & Debug log uploads

[gre.zip](https://github.com/SoftFever/OrcaSlicer/files/14562452/gre.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

igiannakas, 2024-03-11:
Check your printer settings under motion ability. Is it set to over 500mm/sec acceleration?

Normal1no, 2024-03-11:
> Check your printer settings under motion ability. Is it set to over 500mm/sec acceleration?
[image]
[image]

velocity is normal, accel to. but this isn't applicable to my issue

igiannakas, 2024-03-11:
No I mean in orca - in the edit printer settings…

Normal1no, 2024-03-11:
> No I mean in orca - in the edit printer settings…

[image]
yea

Normal1no, 2024-03-11:
[image]
I found a problem, but idk how to solve it

igiannakas, 2024-03-13:
Set your max accel for extruding and retracting to 7k and it should work

[image]

In the latest version (2.0 beta) you'll get appropriate warnings like the below:
[image]

henrivdr, 2024-03-23:
@igiannakas 
The new warning does not make sense. Why does it compare the Extrusion Acceleration with the other acceleration(s)?

Bumping it to 7000 is just a workaround, right?  Defeating the purpose of that limit.
Won't it cause issues with the extrusion?

igiannakas, 2024-03-23:
The label should say maximum acceleration while extruding

henrivdr, 2024-03-24:
Ahh, that makes sense.

henrivdr, 2024-03-24:
Ahh, 
I also looked at the source and it emits M204 for machine_max_acceleration_extruding.
When I looked at the two values _extruding and _retracting, I wrongly assumed they were for the extruder **_motor_** itself.

The gcode docs explains it properly.
https://marlinfw.org/docs/gcode/M204.html
P<accel>]	
Printing acceleration. Used for moves that **_include_** extrusion (i.e., which employ the current tool).
[R<accel>]	
Retract acceleration. Used for extruder retract / recover moves.

closed 2024-06-30 by github-actions[bot]