OrcaSlicer/OrcaSlicer #4422: "Keep" and "Discard" are confusing options in the "unsaved changes" dialog.

Raketemensch, 2024-03-11:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The text says, "Would you like to keep these changed settings (new value) after switching presets?"

The table in the dialog shows "Old Value" and "New Value," and the button choices are "Transfer" and "Discard," which don't make sense in this context.

The buttons should say "Use New" and "Keep Old" or something more explicit.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

The buttons should say "Use New" and "Keep Old" or something more explicit.

They should leverage the text that is already in the dialog.

mtmello2003, 2024-03-13:
I totally agree.

Eldenroot, 2024-03-13:
Yes

xTremePower, 2024-04-08:
totally agree ..
It made me errors many times ... and saved what I don't want to save ...

nmcandrews, 2024-06-03:
100% agree.

SoftFever, 2024-06-03:
It should be more clear/consistent in 2.1.0 beta now
[image]

ShayGus, 2024-09-06:
This is sooooooo confusing.
What am I transfering new to old, old to new?
What am I discarding of? The new value, the old one.
I never quite sure what's going on the re when I press either.
As a user I don't have to need to go and look at the code to understand a simple dialog.
My UX is that I totally and absolutelly confused by this dialog.

sdylewski, 2024-10-31:
I'd really like some more help on this screen also.  My wife and kid and I all use the same slicer and we're all just playing roulette every time we click one of the buttons because we don't understand it.  It shouldn't be that way.  This is just not obvious to me what each option does.

PitBull440, 2025-01-01:
I agree completely.  It's almost like this was written as a cruel joke to confuse people purposely.    

How about putting a checkbox above or below each column so we can clearly select which settings we are accepting?

And what does the "Save" button do?

adobni, 2025-01-22:
Current options are confusing. Am I overriding the defaults? or is it the project? what is transferred to what?

closed 2025-01-28 by SoftFever

wickedlemon2, 2025-01-28:
> Please read the description in the dialog, it's pretty clear already.
> 
> This dialog popup when you switch presets and there are unsaved changes. So these options are: **Transfer** (changes to new preset) **Discard** (changes) **Save** changes into current preset then switch to new preset

If it's pretty clear already why did you need to append additional descriptions in your post?

TheSiskoDax, 2025-02-03:
> Please read the description in the dialog, it's pretty clear already.
> 
> This dialog popup when you switch presets and there are unsaved changes. So these options are: **Transfer** (changes to new preset) **Discard** (changes) **Save** changes into current preset then switch to new preset

This is the question: If I am downloading a .3mf from makersworld and I want to use the specific preset values provided by the designer, which option do I choose?

disconn3ct, 2025-02-04:
> Please read the description in the dialog, it's pretty clear already.

Please read the messages in the ticket, it is pretty clear that new users do not understand that dialog.

Now that that is out of the way, can we agree that the dialog phrasing is not perfect and could be improved?

xTremePower, 2025-02-04:
I'm a long term user and it's still not clear at all ...

You lose everyone with the save/transfer things and also discard ...

If i'm on a project, and importe a 3mf with settings, 
What will happens if I Discard ? discard what ? the settings into the 3mf ?
What will happens if I SAVE ? Save into what ? into the profile actually on screen, or save the settings of the 3mf ?
What will happans if I Trasfer ? It will tranfer settings from the 3mf to into the actual settings ? in this case what the difference between save and transfer? 
If I already have the preset saved in my config settings, but I enabled one other one before opening the 3mf it will still create a new preset or use the saved one ?!

Many times I make button error, and I need to think about to push the orange rollback circles in printer, filamen and config windows before saving myself or I will trash my actual good profile .....

MxBrnr, 2025-04-25:
Apparently I am the only person who has thought the implementation of this dialog has been completely self-explanatory from the beginning. 🤷 Admittedly, "Transfer" is better and less confusing than "Keep," but beyond that the 3 option are clearly different and seem quite obvious, with the wording being adequately consistent with what is commonly used in software and other digital applications.

klwemu, 2025-04-25:
It's totally confusing and it is not enogh to change some text labels. 
The UI should be redesigned and to be more intuitive.

PitBull440, 2025-04-27:
When you Transfer the changes to the newly selected preset, does it Save or Discard the changes to the preset that _was_ selected?

MxBrnr, 2025-04-27:
Discards, because you did not save them to that profile before switching.

nilesjohnson, 2025-05-02:
I don't think 2 min of self-experimentation clarifies this, nor do I think saying so is a helpful way to proceed.  

I found this issue while internet searching for some explanation.  I have a project where I've tweaked a bunch of settings just for those particular objects.  When I try to quit Orca Slicer, I'm presented with this dialog.  I'm not trying to change presets.  I don't want to lose the settings I've dialed in for this particular project, and I also don't want these settings being copied over to my other projects.  I can't imagine this is an unusual situation.

Also, please clarify about #9477: can this issue be reopened and that one closed as duplicate?

MxBrnr, 2025-05-03:
> I don't think 2 min of self-experimentation clarifies this, nor do I think saying so is a helpful way to proceed.
>
> I found this issue while internet searching for some explanation. I have a project [...]

Then you need to save those settings as a new profile, or save the project 3MF, before changing or exiting

nilesjohnson, 2025-05-05:
> Then you need to save those settings as a new profile, or save the project 3MF, before changing or exiting

Thanks.  If the dialog isn't going to change, how about adding a section on the wiki page so that anyone who is still confused can find extra explanation there?  A starting point would be [this earlier comment](https://github.com/SoftFever/OrcaSlicer/issues/4422#issuecomment-2618329228).

>This dialog popup when you switch presets and there are unsaved changes.
So these options are:
Transfer (changes to new preset)
Discard (changes)
Save changes into current preset then switch to new preset

While doing this, someone could clarify whether any of the buttons saves the current 3mf, or that has to be done separately.

Depending on what's best, I'd be happy to
- open a new ticket for this, 
- send a PR with draft changes for the wiki, and/or
- leave it to someone else who knows better how to do these things

MxBrnr, 2025-05-07:
> While doing this, someone could clarify whether any of the buttons saves the current 3mf, or that has to be done separately.

The 3MF must be saved separately.

klwemu, 2026-09-25:
What, it is "Closed" - unsolved .... I can't belive.

nilesjohnson, 2026-09-25:
There is now a section of the wiki called [Transfer or Discard Changes popup dialog](https://www.orcaslicer.com/wiki/general_settings/transfer_discard_changes.html) that explains what the buttons do, based on some of the discussion above.  Perhaps a link to that explanation could be added.

Would the maintainers be open to this?  If so, I can make a new issue and make a PR that adds the link.