OrcaSlicer/OrcaSlicer #4432: Post Processing Script not Applied to "Print ..." or "Send", Only to  "Export G-Code File"

CCS86, 2024-03-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0 Beta

### Operating System (OS)

Windows

### OS Version

10

### Printer

X1C

### How to reproduce

Add a post processing script to the setting in Others.

Export the G-code file, "Print" the file, "Send" the file.

Compare the resulting gcode files.

### Actual results

On the file exported will have been post processed.

### Expected results

I expect that regardless of which output method you use, the output gcode should be identical.

### Project file & Debug log uploads

...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SHoover80, 2024-03-12:
Would this be the same reason or related to the reason that when I put START_PRINT, and END_PRINT for my start and end g-code in OrcaSlicer printer settings and send file directly to my 3D printer over network to print it wont follow my macro's and start the heatup process of the bed or hotend and gives me an error stating SDCARD_RESET_FILE cannot be run from the sdcard when I'm not using a sdcard? And using these Macros ( START_PRINT/END_PRINT) for my start/end g-code works with all other slicers? As a test I took the OrcaSlicer default start/end g-code when setting up printer and inserted that into my START_PRINT/ END_PRINT Macro in my 3D printer macro config file and it wont work either. It's behaving like it's not sending the Start/end macro along with the file when I upload directly to my 3D printer.

JacopoDonnini, 2024-04-01:
I am in desperate need for this issue to be fixed. I added a post processing script to my print configuration that extracts all the useful information and uploads it to an excel spreadsheet which is needed in my workplace to keep track of used materials and costs, but the only way I actually get to do this is by exporting the gcode file, so that the post-processing script is triggered, only to then re-slice the model and send it directly to print. Not only this introduces too many steps for starting a print, but it also clutters my pc with useless gcode files used only to execute the post-processing script.

closed 2024-07-09 by github-actions[bot]