OrcaSlicer/OrcaSlicer #4435: Disabling "Ensure Vertical Shell Thickness" breaks "Don't Filter Out Small Internal Bridges"

mf7120, 2024-03-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Ender 3 S1

### How to reproduce

Disable "Ensure Vertical Shell Thickness" and use "Don't Filter Out Small Internal Bridges -> Limited filtering"

### Actual results

Just by disabling "Ensure Vertical Shell Thickness" no internal bridges will be created to support small top surfaces and they drop down giving bad results. First 2 pictures with EVT OFF. The other 2 with EVT ON -> ALL
[image]
[image]
[image]
[image]

### Expected results

Be able to create internal bridges regardless of ensure vertical thickness turned off or on..

### Project file & Debug log uploads

See layers from 3.36mm to 5.00mm
[EVTON-OFFISSUE.zip](https://github.com/SoftFever/OrcaSlicer/files/14576170/EVTON-OFFISSUE.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

igiannakas, 2024-03-13:
I still see the internal bridges generated correctly when critical only is selected on ensure vertical shell thickness.

[image]

What you're seeing is less solid infill needing support, hence less internal bridging happening.

Dont use "None" on ensure vertical shell thickness, this disables it completely and for your model it won't work.

mf7120, 2024-03-13:
@igiannakas that's the thing, as mentioned internal bridges should be still created even if EVT is set to "None". Prusa Slicer is able to do it. Also for me personally EVT just adds unnecessary material and increases printing time it even handles better with "All" instead of "Critical Only".

Also it is kind of weird now that I see it that the amount of top layers are not being generated with EVT set to "None"

igiannakas, 2024-03-13:
Prusa slicer doesn’t have the evt option. None should not be used for regular printing…

mf7120, 2024-03-13:
@igiannakas exactly. They got rid of that function many versions ago since it didn't serve any purpose and as mentioned it just added unnecessary material and time to any print.

closed 2024-06-20 by github-actions[bot]