OrcaSlicer/OrcaSlicer #4439: Elegoo Neptune 4 Pro thumbnail script error after the update 2.0.0 beta

TochkaIco, 2024-03-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Elegoo Neptune 4 Pro

### How to reproduce

1. Add a model on a plate
2. Slice it
3. Export g-code file/Print 

### Actual results

at this point I get an error after the script is run, this didn't happen with previous versions
[image]

### Expected results

The script should run without errors and should add a thumbnail to the g-code.

### Project file & Debug log uploads

[logs.zip](https://github.com/SoftFever/OrcaSlicer/files/14578526/logs.zip)
[project.zip](https://github.com/SoftFever/OrcaSlicer/files/14578545/project.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

mtc1122, 2024-03-13:
Chances are, the update changed the default back to the incorrect Colpic instead of PNG, and changed the resolutions back to the incorrect thumbnail resolutions.

TochkaIco, 2024-03-13:
Yeah, the update changed it to Colpic 320x320, 160x160;
After the changes, it's back to normal, thank you for the help:)