OrcaSlicer/OrcaSlicer #4440: Assign mouse buttons

MrDB42, 2024-03-12:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I would really like to be able to reassign/swap the mouse buttons for rotate and pan. It's frustrating going from my CAD software and then to Orca Slicer and find that they are opposite controls. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would like to be able to go into "Preferences" and be able to have the choices of:
Pan: Left mouse button or Right mouse button
Rotate: Left mouse button or Right mouse button

Where if you choose one it will swap the other.

### Describe alternatives you've considered

I can't think of any alternative ways of doing this.

ramosglauco, 2024-03-23:
Please, add this functionality. This is present in PS and would be of great help if it were also here.

[image]

Ale11Re, 2024-05-05:
+1
In addition, while viewing model in the prepare tab, the only way to rotate view is by left click, which is also for model movement. It happened too many times that I misplaced model for that. Also, right click and wheel click do exactly the same. That is a waste of buttons.
Letting user assign its preferred modes would be ideal.

gknapp, 2024-05-07:
Similar issue open for PS - open for years too. Would like to see mouse button assignment in preferences please. The default doesn't match CAD software.
https://github.com/prusa3d/PrusaSlicer/issues/4277

csstup, 2024-07-21:
Wanted to pick up a project to help give back to OrcaSlicer, thought this would be a good one to get started on.

I thought it wouldn't be a bad idea to allow each of the buttons to be assignable to an action on drag, so you can set each button to either "Rotate", "Pan" or "None".     

As a mockup, I hacked the preferences panel a bit to give more room for the labels on the left and coded up the logic for the dialog, the preferences defaults and the mouse handling routines.    So far it works great, but the UI needs polishing.    The alignment isn't great, one of the labels doesn't fit, etc.   The labels/options may not make sense and I'm really interested in hearing options/suggestions.

Here's what I have so far:

[image]

This shows the defaults as what Orca would usually have set.    You can change it as you'd like:
[image]

The changes are pretty simple and I can commit it to a branch on a fork for code review if this seems like a viable solution.

Thanks!

metalman1884, 2024-11-13:
> Wanted to pick up a project to help give back to OrcaSlicer, thought this would be a good one to get started on.
> 
> I thought it wouldn't be a bad idea to allow each of the buttons to be assignable to an action on drag, so you can set each button to either "Rotate", "Pan" or "None".
> 
> As a mockup, I hacked the preferences panel a bit to give more room for the labels on the left and coded up the logic for the dialog, the preferences defaults and the mouse handling routines. So far it works great, but the UI needs polishing. The alignment isn't great, one of the labels doesn't fit, etc. The labels/options may not make sense and I'm really interested in hearing options/suggestions.
> 
> Here's what I have so far:
> 
> <img alt="image" width="370" src="https://private-user-images.githubusercontent.com/8303565/350741896-9055f560-e7a6-4e5f-a340-16d91b771123.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE0ODkyOTEsIm5iZiI6MTczMTQ4ODk5MSwicGF0aCI6Ii84MzAzNTY1LzM1MDc0MTg5Ni05MDU1ZjU2MC1lN2E2LTRlNWYtYTM0MC0xNmQ5MWI3NzExMjMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTExMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMTNUMDkwOTUxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDg5OTVjOGEzM2E1OWMzNDBhZDJhMTRmNGVkZDg2OGJjZDY3M2Y0N2NiYTljNTFkMjNmNjNiOWIzYmY0ZjJiZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.JFA7HaN5ULXuhLX_OKkfMDGuT2cvB1G_bac_MjAJcR8">
> This shows the defaults as what Orca would usually have set. You can change it as you'd like: <img alt="image" width="370" src="https://private-user-images.githubusercontent.com/8303565/350741984-3495a214-230c-49ca-a4bd-4796b21b0887.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE0ODkyOTEsIm5iZiI6MTczMTQ4ODk5MSwicGF0aCI6Ii84MzAzNTY1LzM1MDc0MTk4NC0zNDk1YTIxNC0yMzBjLTQ5Y2EtYTRiZC00Nzk2YjIxYjA4ODcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTExMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMTNUMDkwOTUxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZjEwYTI0Mzk0N2QwYjk0ZTdmMzk5MTA0MmJkODc0MzdjZWRmNDY0MjFhMTYwZDMzY2E5NTZmYjJjYTc3OGJkOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.qwg50kZEYr4wcJUcI322h4R6JCubKjO5ZthBo0SzA-Q">
> 
> The changes are pretty simple and I can commit it to a branch on a fork for code review if this seems like a viable solution.
> 
> Thanks!

@csstup 
This feature needs to exist on every slicer! Someone already commented but there are always misplacements because the same button does both move part and pan camera, and it's really frustrating to go between your favorite CAD and Orca and having to remember which button does what. Any updates on if this could be implemented on the next release?

JaneSmith, 2024-12-10:
I'd like this feature. I much prefer to have left click select, middle click pan, and right click rotate. I believe this is referred to as "TinkerCAD" style, although I've never used that.

Creality Print does it this way and it feels much nicer to use for me since it matches my CAD software.

travisterrell, 2025-03-24:
> The changes are pretty simple and I can commit it to a branch on a fork for code review if this seems like a viable solution.
> 
> Thanks!

Hi there, @csstup! Would you mind sharing your branch where you made these changes? I'd like to try to merge them into latest locally so that I can finally solve my frustrations. (And maybe if it gets polished enough, I'll write some unit tests and you could make a PR with everything and we could get the ball rolling on getting this in place. Many people would love you for it!)

But either way, if you wouldn't mind sharing, I would be eternally grateful. Thanks in advance!

eyeonus, 2025-06-08:
> The changes are pretty simple and I can commit it to a branch on a fork for code review if this seems like a viable solution.

Please do!

eyeonus, 2025-06-08:
> The changes are pretty simple and I can commit it to a branch on a fork for code review if this seems like a viable solution.

@csstup Please do!