OrcaSlicer/OrcaSlicer #4441: Mispell

thefluxplux, 2024-03-12:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

 Under Quality tab --> Wall and surfaces --> Wall loop direction --> Description has a misspell :D

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

 Under Quality tab --> Wall and surfaces --> Wall loop direction --> Description has a misspell

cochcoder, 2024-03-12:
What is the missed spelt word? Do you mean Walls printing order?

cochcoder, 2024-03-12:
Found it

PR #4444 (2024-03-13): Minor spelling & grammatical fixes
Closes #4441, #4447
closed 2024-03-13 by SoftFever