OrcaSlicer/OrcaSlicer #4442: Changing world coordinates not working for added parts

gasprints, 2024-03-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

macOS

### OS Version

MacOS 14.3.1

### Printer

Bambu Lab A1 Mini, P1P, P1S, Creality Ender 6 (Klipper)

### How to reproduce

Load a model, add a part, try to type world coordinates to move the added part.

https://github.com/SoftFever/OrcaSlicer/assets/128255447/d24ba68a-d8b3-43c8-b1c6-8ca03b12e2fb

### Actual results

Trying to move it to a defined coordinate moves it in relation to the previous coordinates instead of to the defined coordinate

### Expected results

Should move exactly to the coordinate

### Project file & Debug log uploads

[Broken.zip](https://github.com/SoftFever/OrcaSlicer/files/14578946/Broken.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

mf7120, 2024-03-12:
I can confirmed I noticed this today, it also applies to any kind of modifier

solidus0079, 2024-03-18:
Noticed this happening with Negative Cylinder Shapes, was about to submit an isseue on it.
[image]

kleinwareio, 2024-03-24:
Happens on 1.9.1 as well

GrumpyOldDude, 2024-03-29:
This has been driving me mad for the last 2 weeks.. Finaly found this post, thought it was something in my settings but i gues its a bug. Installed V 2.0.0 RC release but that does not fix it. Had to go back to installing version 1.9.0 to get rid of this problem. Thanks to the people posting here cause i was losing my mind lol.

gchwelos, 2024-04-01:
Happens on Windows version too, Windows 10. Release 2.0.0. When I type in specific coordinates to Move a modifier (cube), the resulting coordinate is random; not what was entered. First time specifiying Move coordinates usually works; after that results are random. Must delete modifier and start again.

appleimperio, 2024-04-09:
Exact same issue with added part the coordinates are off

rmcardle, 2024-04-09:
I am seeing this issue as well. Very frustrating because I have no way of accurately positioning a modifier for specific layers.

These issues https://github.com/SoftFever/OrcaSlicer/issues/3636 and https://github.com/SoftFever/OrcaSlicer/issues/4845 look to be the same or related.

beanzen, 2024-04-28:
I have the same issue on 2.0.0 on Mac ARM64 as well. It's maddening because I can't work with anything. Downgrading to 1.9.0

solidus0079, 2024-04-28:
> I have the same issue on 2.0.0 on Mac ARM64 as well. It's maddening because I can't work with anything. Downgrading to 1.9.0

It's been fixed on the 2.10 daily build

mtbikemarc, 2024-06-01:
Please fix this problem. It is very frustrating. :-)

solidus0079, 2024-08-31:
It's been fixed, Mr. Robot.  Someone with authority just needs to mark it closed :-)

closed 2024-12-09 by github-actions[bot]