OrcaSlicer/OrcaSlicer #4443: Problems with using Macro for Start/End G-code

SHoover80, 2024-03-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

V2.0.0-Beta

### Operating System (OS)

Windows

### OS Version

Version 23H2 (OS Build 22631.3235)

### Additional system information

Gigabyte B650 Aorus Elite AX Mobo
AMD Ryzen 7 7800X3D
Corsair Vengence 2 x 16GB DDR5 7000Mhz
Gigabyte RTX 4070 

### Printer

Heavily modified Ender 5 Pro. 220 x 220 x 500, Mercury 1 Zero G Core XY, Dual Z motor/belt, Manta M8P/Raspberry Pi CM4, Klipper Version: v0.12.0-117-g71604b71 with EBB36 CAN. 

### How to reproduce

Using START_PRINT AND END_PRINT macro in OrcaSlicer printer settings for start/end G-Code. 

### Actual results

It'll home then give me multiple warnings about being unable to extrude due to temps being below specified temp range then gives me an error stating SDCARD_RESET_FILE cannot be run from the sdcard when I'm not using a sdcard. I sent the file directly through network from OrcaSlicer to 3D printer. 

### Expected results

As specified by START_PRINT macro it should turn on webcam LED, Heat up hotend/bed, Home, check bed level on 3x3 grid, run a prime line then start printing.  as it does when using START_PRINT/END_PRINT for start/end g-code settings for Cura or SuperSlicer. 

### Project file & Debug log uploads

[OrcaSlicer issue.zip](https://github.com/SoftFever/OrcaSlicer/files/14578916/OrcaSlicer.issue.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

I've tried with the START_PRINT/END_PRINT macro's I've been using and worked with other slicers. Someone suggested in discord it may contain something not compatable with OrcaSlicer related to the mesh, Which didn't make sense but out of desperation I tried but this didn't resolve the issue. Out of further desperation getting something that should be simple to work but has been nothing but problamatic,  I took the default (working) start/end G-Code provided by OrcaSlicer printer configuration and set that as my START_PRINT/END_PRINT macro in my 3D Printers macro config file then entered the START_PRINT/END_PRINT as my start/end G-code in OrcaSlicer printer config settings knowing the macro is directing it towards the G-Code provided by OrcaSlicer itself It should work. But I ended up with the same results. It completely ignored or failed to include the start macro with the sliced file it sent to 3D printer. It homed, then showed the same error about temp too low to extrude because it made no attempt to heat up, and complaing about SDCARD when again, The file was sent over network. 
OrcaSlicer printer settings:
[image]

My original Start/End Macro that works with other slicers: 
[image]

How I entered the Macro in OrcaSlicer printer settings:
[image]

I've even tried these macro's for START/END :
[image]

Which is the default start/end provided by OrcaSlicer
[image]

Working start/end G-code setting in Cura:
[image]

igl-x2, 2024-03-13:
When loading a project, it will also load the used start code. Not your current code.
Click on reset settings to change to your New code

igl-x2, 2024-03-13:
https://www.klipper3d.org/Slicers.html

SHoover80, 2024-03-13:
I'll give that a read, Thanks for the response.

SHoover80, 2024-03-13:
@igl-x2, Ok, I read it. And it failed to provide any new information relevant to my issue. It just confirms what I've said. That I should be able to use START_PRINT/END_PRINT to run my Macro's that are configured in my printer settings. But it's not working with OrcaSlicer. Hence, the reason I'm here and not over on Klippers site requesting assistance from Klipper.  It did say to choose Marlin as the G-Code flavor since many slicers dont give the option to choose Klipper.  I've tried that during one of my many troubleshooting attempts over the past 4 days and it made no difference. It showed some G-Code/Macro examples but I know the macro's I've been using are correct and work with my printer because I've been using them for about a year with no issues with other slicers.  So glad to know I've spent all that time to provide detailed info about my issue, screen shots, logs and the project file for someone to glance over the title and post a basic site thinking they're contributing.

vgdh, 2024-03-13:
You need to pass temperatures as a parameters:
START_PRINT BED_TEMP=[first_layer_bed_temperature] EXTRUDER_TEMP=[first_layer_temperature]

SHoover80, 2024-03-13:
@vgdh, Would you mind elaborating on that a little more? Do I just throw that line in the START_PRINT macro?

vgdh, 2024-03-13:
> @vgdh, Would you mind elaborating on that a little more? Do I just throw that line in the START_PRINT macro? 
> 

throw that line in the orca slicer. 
you already have these parameters in your klipper macro

SHoover80, 2024-03-13:
@vgdh, Got ya, I tried that and it worked. Thank you very much.

vgdh, 2024-03-13:
> @vgdh, Got ya, I tried that and it worked. Thank you very much.

You are welcome 😉 . 
please close the issue if it's solved.

closed 2024-03-13 by SHoover80

SamBytess, 2024-07-01:
END_PRINT needs to have the if statements included!  Can you not do it without the if statement?