OrcaSlicer/OrcaSlicer #4447: Counter bore mispelled

mtc1122, 2024-03-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Elegoo Neptune 4 Plus

### How to reproduce

1.  Go to "Quality"
2. Scroll down to "Bridge counterbole holes"
3. Notice that "counterbore" is misspelled as "counterbole" (including in one spot in the tooltip).

### Actual results

Notice that "counterbore" is misspelled as "counterbole" (including in one spot in the tooltip).

### Expected results

counterbore

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

[image]

cochcoder, 2024-03-13:
This issue should now be fixed

cochcoder, 2024-03-19:
Please close this issue if Counterbore isn't misspelled anywhere else

closed 2024-06-26 by github-actions[bot]