OrcaSlicer/OrcaSlicer #4450: Tree supports "floating branches"

Silent6assassin, 2024-03-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0

### Operating System (OS)

Windows

### OS Version

Win 10

### Printer

K1 and V400

### How to reproduce

Loaded file and used default tree supports on default style with on build plate only. Happens on a lot of models that are only a few mm off of the build plate that use many tree supports for underneath

### Actual results

[image]
First layer creates brim for supports, next makes the base for a trunk, but after a few layers, it starts to make branches that have nothing to connect to and are "floating" thus creating a mess of unconnected supports and the base of the model unsupported or bad quality if it somehow catches.

### Expected results

Would expect each section would have its own trunk/base to connect on the bottom layers and not auto create in mid air, eventually connected all branches to bottom of model.

### Project file & Debug log uploads

[Gengar Support issue.zip](https://github.com/SoftFever/OrcaSlicer/files/14581577/Gengar.Support.issue.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

Using almost all default settings across 2 printers

Edzi11, 2024-03-13:
have had this happen multiple times and the only solution for me was to add my own supports via fusion 360. wish they would improve the supports generation

Silent6assassin, 2024-03-13:
I have attempted to on a few models, but have noticed on some of the models I have it crashes fusion since there are too many triangles. Along with I have encountered on one model, however the designer made it, editing it seemed to mess up the original model, so it seems while this is a possibility, it doesn't always work out. I also Tried paint on supports and didn't seem to make any difference on that either.

Edzi11, 2024-03-14:
> I have attempted to on a few models, but have noticed on some of the models I have it crashes fusion since there are too many triangles. Along with I have encountered on one model, however the [...]

Ya it's a pretty big hassle to do to be honest. This has been an issue for quite awhile though so not sure if they'll ever get to it.

closed 2024-06-21 by github-actions[bot]