OrcaSlicer/OrcaSlicer #4452: New scarf seam gives odd errors on specific models and orientations when Arachne is enabled

Edzi11, 2024-03-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0 beta

### Operating System (OS)

Windows

### OS Version

windows 11

### Printer

bambu x1c

### How to reproduce

1.  load model and auto-orient 
2. using standard .2 profile have scarf seam enabled and arachne wall enabled 
3. gives error saying gcode goes beyond max height and shows last layer is 925k mm tall when in reality its 156mm tall
4. if you place the model facing down it does not give the error
5. There is also a very similar model with no issues and the plate is labeled as no issue 

### Actual results

[image]

1. as shown on the screenshot getting error of going beyond max height for last layer. when looking at the right side where the model height is shown it gives an insanely high number for the model height
2. only happens when arachne and scarf seam are enabled and the model is standing, wont happen if the model is facing down
3. Plate labeled No Issue doesn't appear to face any errors 

### Expected results

1. Should be able to have Arachne and scarf seam enabled at the same time with no issue regardless of the model or orientation

### Project file & Debug log uploads

[Error project.zip](https://github.com/SoftFever/OrcaSlicer/files/14583630/Error.project.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

vgdh, 2024-03-13:
Attached file cannot be download. Please attach the file of the project.

Edzi11, 2024-03-13:
> I confirm, there is some strange behavior. But if I change the printer setting to another printer it's works for some reason.

Glad you were able to confirm the issue. Were you able to download the 3MF containing the error project?

vgdh, 2024-03-13:
> Glad you were able to confirm the issue. Were you able to download the 3MF containing the error project?

Yes I checked it with **error project** file

closed 2024-06-20 by github-actions[bot]