OrcaSlicer/OrcaSlicer #4456: Port Automatic FDM support painter from PrusaSlicer

MrAlex1124, 2024-03-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Not necessarily a problem, more of a convenience feature, but a very nice feature in the PrusaSlicer is automatic painting.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Automatic FDM support painter was added in PrusaSlicer version 2.6. I would like to see this feature included in OrcaSlicer. It makes painting the supports very easy. One push of a button and all supports are in perfect place.

Eldenroot, 2024-03-13:
I think that @Noisyfox had tried to implement... but not successful :(

Eldenroot, 2024-06-12:
B

closed 2024-09-18 by github-actions[bot]