OrcaSlicer/OrcaSlicer #4457: Can move items on a locked plate

tlhintoq, 2024-03-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0-dev

### Operating System (OS)

Windows

### OS Version

Win10x64

### Additional system information

Unrelated to bug

### Printer

Unrelated to bug

### How to reproduce

1. Multiple items across 2 or more plates
2. Lock plate 1
3. Select plate 2 as the current working plate
4. Select-All {Control + A}
5. Move the objects

### Actual results

**ALL** of the object on all of the plates move - including the locked plate

[image]

### Expected results

Only the unlocked plate(s) should be affected.

Just a thought... Maybe `Control+A` should only select on the selected plate.  `Control+Shift+A` could be a less common `Select across all plates`.

### Project file & Debug log uploads

Unrelated/unrequired for this bug

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

[image]

tlhintoq, 2024-04-15:
Keep alive.  Problem still exists in current 2.0 nightly as of 15apr2024

New job
Add primative: Torus
Lock plate
Grab and move

So what does the lock even do?

tlhintoq, 2024-05-10:
Bot nudge.

tlhintoq, 2024-08-09:
"Bot don't close me now" - This is still very much needed.

disconn3ct, 2024-08-09:
Stalebot is here to say that the maintainers aren't going to triage this and we should go away.

tlhintoq, 2024-08-10:
A bit pessimistic - but I know it can feel like that at times.
If something *can't* be addressed for a reason like... its too deeply ingrained from the master version that Orca inherits from (I think that's Prusa slicer or Bambu Slicer) then someone should just say that.  We'd all respect it.
But otherwise - since so much work is from volunteers I think it mostly means that nobody feels comfortable tackling the matter after looking at the code for that feature.  I've seen some horrendous gerry-rigged code that someone looks at and says
> Oh helllllll no.  I'm not touching that beast.

closed 2024-11-16 by github-actions[bot]