OrcaSlicer/OrcaSlicer #4459: Custom Profiles lost on update

spikeon, 2024-03-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Elegoo Neptune 4 Max

### How to reproduce

Update from 1.9.1 to 2.0.0-beta with a childlike glee about being able to try out scarf seams
Go to print
Find that all of your Custom Printer, Filament, and Process profiles have all been deleted

### Actual results

all of your Custom Printer, Filament, and Process profiles have all been deleted

### Expected results

My profiles should still be there after the update

### Project file & Debug log uploads

Not Applicable

### Checklist of files to include

- [ ] Log file
- [ ] Project file

spikeon, 2024-03-13:
Update:  This only appears to have effected my imported Machine profile and it's associated FIlaments and Processes.

DrYug, 2024-03-13:
Hello, i have the exact same problem. When i use Orca 1.9.1 in my laptop all work fine. but in my desktop all of my profiles are gone

mlewis-everley, 2024-03-18:
I have got this too, using the Linux client. Even if I try going back to a previous version my profiles are lost (possibly because all profiles are on a shared location)?

mlewis-everley, 2024-03-18:
OK, I just realised, for me, the upgrade removed all my default profiles. All my printers run custom settings built on a system preset version.

I had to manually re-add all the system default printers that were removed, and now all my custom presets are back.

tobilab, 2024-03-22:
Check if [this bug](https://github.com/SoftFever/OrcaSlicer/issues/4406) is related and check whether you have selected the nozzle diameter the filament profiles where created for.

jmvaughan2, 2024-03-26:
I believe I have this issue and it doesn't look to be related to the bug @tobilab mentioned. In my case printer, filament, and process profiles were all missing for my custom printers but not my bambu A1. Files are still there as others mentioned. If I close out Orca and reopen it soon after they show back up, but it switches to the bambu being the last printer used.

jmvaughan2, 2024-04-03:
comparing log from when it doesn't load them to when it does. I get the following when it doesn't load. Verified the inherits are there just to be sure though when it works it doesn't give an error about fining them.

[Thread 0x00004a7c]:Slic3r::PresetCollection::load_user_preset, can not find inherit preset for user preset

jmvaughan2, 2024-06-15:
This is still present in 2.1.0-Beta

closed 2024-09-21 by github-actions[bot]