OrcaSlicer/OrcaSlicer #4461: "Access Violation"

ThinkBigPrintHuge, 2024-03-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

V2.0.0 Beta

### Operating System (OS)

Windows

### OS Version

Windows 11 Home

### Additional system information

CPU: i9-12900h
GPU: RTX 3070Ti (Laptop)
Ram: 32g

### Printer

re:3D Gigabot 4

### How to reproduce

Enable "Small Area Flow Compensation" and click slice.  Sometimes the error message come up, sometimes Orca completely closes. Seems dependent on what model is loaded. 

### Actual results

[image]

The above error message is received.  The first time this happened I had to uninstall and reinstall Orca. If I slice with "Small Area Flow Compensation" now, I can just close Orca and reopen it and it works. A bunch of random error messages kept popping up which led me to reinstall.

### Expected results

Not sure. I assume Orca shouldn't crash.

### Project file & Debug log uploads

I can't get anything to save after the error, so I can't upload anything here showing the problem.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bugpwr, 2024-03-13:
I have this crash log from just now:

Exceptions:
Exception Code :c0000005 ACCESS_VIOLATION
Exception Flag :0x0 NumberParameters :2 
Param 0 :0x0 
Param 1 :0xad 
Context :000000BA6ACD1680 
ContextFlag : 0x10005f, EFlags: 0x10206 
Fault address:  0x49710FA9 0x1:0x9CFFA9 C:\Program Files\OrcaSlicer\OrcaSlicer.dll

Show CallStack:
7ff849710fa9:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ff8496edf59:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ff84a843329:hid_write [Unknown file][-1]
7ff84a8a6408:hid_write [Unknown file][-1]
7ff8496c725e:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ff848e42c93:Unknown [Unknown file][-1]
7ff848ea1de4:orcaslicer_main [Unknown file][-1]
7ff6b1c45a45:Unknown [Unknown file][-1]
7ff6b1c45f06:Unknown [Unknown file][-1]
7ff92155257d:BaseThreadInitThunk [Unknown file][-1]
7ff9236eaa58:RtlUserThreadStart [Unknown file][-1]

Logical Address:
0x49710FA9 0x1:0x9CFFA9 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x496EDF59 0x1:0x9ACF59 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4A843329 0x1:0x1B02329 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x4A8A6408 0x1:0x1B65408 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x496C725E 0x1:0x98625E C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x48E42C93 0x1:0x101C93 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x48EA1DE4 0x1:0x160DE4 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0xB1C45A45 0x1:0x4A45 C:\Program Files\OrcaSlicer\orca-slicer.exe
0xB1C45F06 0x1:0x4F06 C:\Program Files\OrcaSlicer\orca-slicer.exe
0x2155257D 0x1:0x1157D C:\WINDOWS\System32\KERNEL32.DLL
0x236EAA58 0x1:0x59A58 C:\WINDOWS\SYSTEM32\ntdll.dll

Razhang, 2024-03-14:
I am also having a similar issue. 

After enabling "small area flow compensation", attempting to slice results in Orcaslicer to crash without any error message. This occurs with any all the models I have tried so far.

-OrcaSlicer version: V2.0.0-beta
-OS: macOS 14.0
-System: Apple M1 chip, 16GB RAM
-Printer: Ender 3 V3 SE

I will also include the debug log from a recent crash as well as a the project file. It is just a cube I was using to test this issue. 

Hopefully this helps, can provide other info too if needed. 

Thank you!
[Test Cube.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14596156/Test.Cube.3mf.zip)
[debug_Wed_Mar_13_21_56_16_90446.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/14596157/debug_Wed_Mar_13_21_56_16_90446.log.0.zip)

Dragonwolf-Dominus, 2024-03-14:
I have the same issue since I installed version 2.0 Beta.

I was about testing new features, and can't slice anything, even after reintalling and rebooting the whole computer.

Seeing the posts here, I desactivate "small area flow compensation", and I can slice.

So it is this feature that make crash the app. I don't understand why, but this is it.
"Access Violation", but I don't have anything in log events. I guess, it is a try to write directly in memory with fixed locations, instead of a var pointer.

Have all a good day.

Even with theses bugs, thank a lot for your work.

ThinkBigPrintHuge, 2024-03-14:
Getting the same access violation error (sometimes) when using fuzzy skin. 2.0.0-dev
I sliced multiple times with fuzzy skin using the same model without issue. Changed one setting and got the access violation error message. Opened a new instance of Orca and reloaded the model and set the same settings and am now consistently getting access violation or Orca completely crashes.

 Let me know if this should be added to a new Bug Report? 

Thanks for all you do!

IntenseWiggling, 2024-03-30:
I'm having the same issue, and I can easily replicate it with a particular model.

- Retrieve "Tetris_Oco_S.stl" from https://www.printables.com/model/335938-balance-tetris/files, and add it to a build plate in OrcaSlicer.
- Select the object, and mirror it on either the X or Y access.
- Attempt to slice the build plate, and receive an access violation.

Can also replicate it the same shaped object as part of "Tetris_D6_Dice_Insert.stl"

PhilBaz, 2024-03-30:
Just got one too.

closed 2024-07-06 by github-actions[bot]

VstudioLAB, 2024-10-25:
i'm having the same issue

HighSchoolProgrammer, 2024-11-26:
I am having same issue on version 2.1.0