OrcaSlicer/OrcaSlicer #4466: Starts to rotate object instead of going through the layers if mouse cursor leaves the window borders

kr15uk, 2024-03-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-beta

### Operating System (OS)

macOS

### OS Version

macOS 14.4 Sonoma

### Additional system information

M1 Pro 14" MacBook Pro and 2020 Intel MacMini

### Printer

Custom

### How to reproduce

1. Slice any object
2. Open Preview and skim through the layers up and down

### Actual results

If you click and hold but move cursor anywhere within Orca window it still goes through the layers BUT if you move cursor outside of the Orca window border it starts spinning the object.

### Expected results

Tried the same in Prusa Slicer and it keeps going through the layers regardless from cursor location.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

netweaver1970, 2024-03-13:
Duplicate issue, I saw it before. It is/was open for a while already. I guess the new features attract more attention than simple annoyances. Very human...
FWIW, this behaviour on MacOS also annoys me.

greendog99, 2024-03-13:
Looks like it was #1807 which was auto-closed as stale because no one worked on it. It drives me crazy too.

closed 2024-06-20 by github-actions[bot]