OrcaSlicer/OrcaSlicer #4468: Improve manual drawing of supports

mirspan, 2024-03-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Better define the support type according to the location on the model. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Ability to draw both classic and tree supports. 

Two types of supports would then be generated on the model.

DVSVIDEO, 2024-03-14:
In OrcaSlicer, there is the possibility to paint the supports. It can be done with a paint brush or a paint can. Then, you can choose either wood or traditional supports.
I do not understand your request...

mirspan, 2024-03-16:
I want to be able to design and print two types of supports simultaneously in the model.

DVSVIDEO, 2024-03-16:
Now I understand...
If you could adjust the support after adding the modifier, you could adjust not only the type, but also the density and orientation (and everything else) within the printout of a model...
Yes that's a good idea...

closed 2024-06-23 by github-actions[bot]