OrcaSlicer/OrcaSlicer #4469: Device Tab Browser Missing WebGL on Linux

Silky322, 2024-03-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.0 - 2.0

### Operating System (OS)

Linux

### OS Version

Arch/EndevourOS 6.7.9-arch1-1

### Additional system information

Display Card: Nvidia 3090 using dkms drivers
Installed Browsers: Chromium, Firefox, Librewolf (default)

### Printer

Ender 3 Klipper/Mainsail

### How to reproduce

1. Open Orcaslicer (either 2.0 flatpak or 1.9 AUR install). 
2. Navigate to device tab
3. Select Heightmap on mainsail

### Actual results

Instead of heightmap see an error showing that browser does not support webgl. 

[image]

### Expected results

Heightmap shows on Firefox and chromium installed on system. 

[image]

### Project file & Debug log uploads

[debug_Wed_Mar_13_11_44_33_43000.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/14592207/debug_Wed_Mar_13_11_44_33_43000.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

My default browser (librewolf) has webgl disabled, but even after changing default in system settings to a known working browser and restarting orca I'm still getting the same error. Don't have this issue on the Windows portable version even with my default browser being librewolf there as well. Is there a way to specify what browser orca uses in config or start args? I thought it came pre-packaged with a webkit/chromium browser based on it's behavior on windows but I'm not familiar with it enough to say definitively.

JimmyFallin, 2024-04-12:
I'm having the same issue using Orcaslicer with my klipper printer CR-10 SE.  it was working when I first rooted the CR-10 SE but have noticed a few updates for Orcaslicer and Bambu Studio and I'm running Linux Mint 21.3.  I only get this when I'm trying to view the mess data for the bed leveling, everything else seems to be running fine.  I can't try Windows, I don't have a Windows machine left I've converted all 4 to Linux. Bye bye Windows you will not be missed.

JimmyFallin, 2024-04-12:
I just narrowed it down to just an issue seeing the mesh data in Orcaslicer.  I fired up my web browser, Firefox and put the IP address in for Fluid 192.168.0.2:4408 and wala it works.  I can click on the tune tab in Fluid and see my mesh data now.  In a latest update of Orcaslicer they must have disabled the webGL support.  I'm not going back the added support for features is greater than my ability to see my mesh data in Orcaslicer.

closed 2024-07-19 by github-actions[bot]