OrcaSlicer/OrcaSlicer #4470: Move out of range at print start

bbinnard, 2024-03-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0 Beta

### Operating System (OS)

Windows

### OS Version

Windows 11/64 with current updates

### Additional system information

This is not a PC related issue

### Printer

FLSun SuperRacer (delta style printer) 

### How to reproduce

At start of print printhead makes 2 small moves down and then stops with error message "Move out of Range". Same GCode file prints fine with PrusaSlicer. Same results with different GCode files. This problem started with Ver. 1.9.xxx and I have not been able to find a fix. 

### Actual results

Printer stops and returns to home position.

### Expected results

Print should proceed

### Project file & Debug log uploads

[klippylogs.zip](https://github.com/SoftFever/OrcaSlicer/files/14593154/klippylogs.zip)
[Project.zip](https://github.com/SoftFever/OrcaSlicer/files/14593155/Project.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

As I said, PrusaSlicer prints my GCode files OK. Bit Orca fails with "Move out of range" error every time. I've looked at the GCode but don't see anything strange. I thought the bed mesh might be a cause, so I created a new one that looks fine in the viewer, but results are the same.

DVSVIDEO, 2024-03-14:
Check the start script in Orcaslicer...

bbinnard, 2024-03-14:
:This is the startup script I've used since I first started using Orcaslicer:

; filament start gcode
G28 ; home all axes
;G1 Z10 F5000 ; lift nozzle
BED_MESH_PROFILE LOAD=default  ; for Klipper

As I said, I've used Orcaslicer for quite a while - until it just stopped working with the error I reported above.  I agree this is a very mysterious problem.

bbinnard, 2024-03-14:
This is my startup script:

; filament start gcode
G28 ; home all axes
;G1 Z10 F5000 ; lift nozzle
BED_MESH_PROFILE LOAD=default  ; for Klipper

I've used this script for Orcaslicer for any months. But I switched back to Prusaslicer when this problem started. I agree it is quite mysterious. And yes, I realize, now that I've looked at the GCode, I don't need the BED_MESH_PROFILE statement. Could that possible be the cause of the problem?

bbinnard, 2024-03-15:
Nope - commenting out the BED_MESH_PROFILE line in the startup script does not fix the problem. If I am the only one having this problem then that suggests there is something wacked in my system. But I am experiencing no other problems - plus Prusaslicer prints the same STL files OK.

I'm going to see if I can find some sort of GCode verifier online.

bbinnard, 2024-03-20:
I tried experimenting with different values for Printer/Printable Height. but none of them resolved the problem.  I'm pretty sure something in the Orca code got wacked a while ago.

Hydra20vt, 2024-03-27:
Same problem here with my FLSun V400. The start GCode with cura works with no issue.

Here the start Gcode:

 {% set BED_TEMP = params.BED_TEMP|default(60)|float %}
  {% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(210)|float %}
  LED_HOTEND_ON
  G21
  G90
  M82
  {% if printer.extruder.target != 0 %}
  M104 S{EXTRUDER_TEMP}
  {% else %}
  M104 S150
  {% endif %}
  M140 S{BED_TEMP}
  {% if printer.toolhead.homed_axes != "xyz" %}
  G28
  {% endif %}
  G1 Z370 F3000
  G1 X-150 F3000
  M190 S{BED_TEMP}
  M109 S{EXTRUDER_TEMP}
  G1 Z1 F3000
  G1 X-150 Y0 Z0.4
  G92 E0
  G3 X0 Y-130 I150 J1 Z0.3 E30 F2000
  G92 E0
  G1 E-1 F2100

bbinnard, 2024-03-27:
Very interesting indeed. I thought my post would end up being ignored because I was the only person experiencing the problem. But now there are 2 of us - at least. 

I know the S400 is larger (specifically taller) than my SR, but is it really 370 mm tall? That's the first thing I noticed about your GCode - why move to Z370 after already doing a G28? Have you tried commenting that line out and then running the GCode?

closed 2024-03-29 by bbinnard