OrcaSlicer/OrcaSlicer #4471: Fuzzy skin paint-on

Skivaks, 2024-03-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Some parts i print get fuzzy skin where its not needed. Example below.
This is how the part looks like with Fuzzy skin off:
[image]
Here is how it looks with available options.
[image]
[image]
[image]

With this particular part i dont need the fuzzy skin on the post closer to the center of the model, only of the outside of the cap where i grab it.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Would be nice to have a Paint-on fuzzy skin tool like we have with Supports. Brushes - Circle, Sphere, Fill.

### Describe alternatives you've considered

I have tried using a Modifier to block Fuzzy skin generation, but it then generates Fuzzy skin where the blocker part ends. See screenshots below.
[image]
[image]

DVSVIDEO, 2024-03-14:
♥In the first lines there is the line that you checked for existing problems...
This is a very old problem. Neither OrcaSlicer nor BambuSlicer existed when this bug was reported in Prusaslicer.
Numerous suggestions have been made (including by me) to dye the fuzzy skin...

Skivaks, 2024-03-14:
> ♥In the first lines there is the line that you checked for existing problems...

I have only checked Issues on OrcaSlicer. I don't use Bambu or Prusa

DVSVIDEO, 2024-03-14:
many-many problem in orcaslicer too
[image]

PLAmenPRINT, 2024-03-29:
I have requested same thing more than year ago but unfortunately the idea was not supported enough.

https://github.com/SoftFever/OrcaSlicer/issues/269

I hope this option (fuzzy skin as paint) will be available some day :)
It will be very nice and usable.

GeorgeStone25, 2024-04-01:
I would like to second the need for this feature. Fuzzy skin makes parts look so much better in general by disguising layer lines, but for people like me who make multi part functional prints, the fuzzy skin feature is unfortunately useless and no system that tries to guess where the fuzzy skin should apply is going to work. I need parts of prints to be smooth, and I need parts to have the fuzzy skin, or they don't work. 

We have paint on supports, so I am kind of surprised that paint on fuzzy skin isn't a thing. 

Even paint on where you just select a face, would be a massive improvement. That I imagine is much easier to do than free paint brush tool. 

I have also tried modifiers like OP, but it's so temperamental and a very frustrating way to do it. It often just adds fuzzy walls randomly in the middle of parts.

jooslouw, 2024-05-20:
Up vote for this request. Paint-on will be great, but in most cases, selecting a face will be perfect.

Also, the problem with modifiers is that if you change your STEP or STL and reload, sometimes the modifiers are not aligned anymore, and you have to go through the painstaking process of realigning the modifiers. Painting or selecting faces will be much easier.

DARKFiB3R, 2024-07-04:
This would be a very welcome feature

simpat1zq, 2024-07-08:
I would love to see this feature.

ph0tic, 2024-07-21:
you have no idea how insanely wanted/needed this is. injection molding does the same thing to hide imperfections.. this is the perfect way to do it with fdm

Adenosina77, 2024-08-03:
Check this out: https://forum.bambulab.com/t/fuzzy-skin-all-or-nothing/5154/5

If you go to Objects, right click and "Add modifier" whatever suits you. Then you select this geometry and remove the fuzzy within. 

FINALLY I can print fuzzy wherever I want. 
I Had to share this with you.
[image]
[image]

ph0tic, 2024-08-04:
Nope don't want that. Creates an additional wall inside of the print for
the fuzzy skin. Really annoying. And then if you have a really curved
surface  good luck with simple modifiers. You can add a copy of the
original model as a modifier but with curves.. resizing correctly to hit it
just right and maybe not make an extra wall inside (sometimes works)  is
horriblly tedious.

So if you are making a square or a cylinder, great. If you are trying to
make something for ppl that abhor later lines...S O L

This is something that's been asked for from slizers for YEARS. I think
there's only one that can maybe do it (the one that you can imprint
designed on the faces).

Pls slicer gods, just make this happen!

TylonHH, 2024-08-18:
> If it is solved in PrusaSlicer, or solved in SuperSlicer, it will be taken over by OrcaSlicer. But there, despite twice as many reports, no results. Either it is very difficult to solve this problem or the developers don't care....
> 
> read this: #2647

It's solved in prusa slicer with 

> You can select to apply Fuzzy skin only to **External perimeters** or to **All perimeters**.

https://help.prusa3d.com/article/fuzzy-skin_246186

DARKFiB3R, 2024-08-19:
> It's solved in prusa slicer with
>
>
> https://help.prusa3d.com/article/fuzzy-skin_246186

That is not solved in the way we are requesting.

We want to be able to paint or fill specific areas with fuzzy skin, just like you can when painting colours.

DBT85, 2024-11-03:
Adding my vote. Sometimes adding modifiers to complex geometries (especially since I can't find a way to change a parts opacity in the prepare view) where just being able to paint on fuzzy or block fuzzy would be really useful!

DARKFiB3R, 2024-11-06:
I really hope this is being considered.

As DVSVIDEO said, maybe it's not as easy to impliment as us non-coders might imagine. I think the developers (at least 1?) do care though, how could they not? It would be such cool feature.

As would the ability to emboss an image/texture onto the surface of a model, like IdeaMaker's slicer.

TylonHH, 2024-11-06:
BONUS Feature: instead of fuzzy skin also a svg with the texture would be a game changer. So you can draw any texture like leather, fur, zebra, waves, etc on the surface.

amckimm, 2024-11-23:
Agreed with all of the above. Paint for many different functions: supports, fuzzy skin, textures, decals, text, embossing. Also ability to save paint template to transfer to other objects.

simpat1zq, 2024-11-25:
Just a heads up that this was released in the PrusaSlicer Alpha 5 days ago. Also, I could be wrong but the PrisaSlicer Alpha release seems like it is saying that OrcaSlicer fixed the issue that OP mentioned in the first post regarding the modifiers printing fuzzy skin at the edge of the modifier block.

ParasitKegel303, 2024-12-26:
+1 Porting this feature from PrusaSlicer would be great.

TheOverpassArsonist, 2024-12-30:
> BONUS Feature: instead of fuzzy skin also a svg with the texture would be a game changer. So you can draw any texture like leather, fur, zebra, waves, etc on the surface.

I definitely agree (as far as I'm aware only ideamaker has this feature currently, which really sucks because it's not even just an aesthetic thing! You can add a pattern of ridges to your part to make it stiffer without adding more plastic, which means doing this is actually a great way to stop warping on larger parts!) but this should definitely be it's own issue.

also +1 on getting paint-on fuzzy skin ported

TylonHH, 2024-12-30:
> also +1 on getting paint-on fuzzy skin ported

Then check this out: https://youtu.be/cNkHfydnUCI

tlhintoq, 2025-02-02:
Fuzzy skin texture painting tool is clearly a desired item.  Just sayin'

https://github.com/SoftFever/OrcaSlicer/issues/7936
https://github.com/SoftFever/OrcaSlicer/issues/4471
https://github.com/SoftFever/OrcaSlicer/issues/3330

FelipeRenault, 2025-02-09:
I've been using this for both top surface fuzzy skin and paint-on fuzzy skin:
https://github.com/TengerTechnologies/Fuzzyficator

There's a video about it too:
https://www.youtube.com/watch?v=cNkHfydnUCI

It's worth noting it still have the extra walls problem, though, since it's pretty much tricking the slicer into thinking it's a color change

JimmyFallin, 2025-03-07:
I would also love to see the "Fuzzy Skin" paint option as trying to use modifiers sometimes is cumbersome and just doesn't work.

Thanks FelipeRenault for posting your solution to our request I am going to give this a shot.

arekm, 2025-05-04:
Wouldn't "paint modifier" be a better, generic option - where area which you painted became a modifier of some depth? That would allow different modifications and not only fuzzy skin.

closed 2025-07-18 by Noisyfox