OrcaSlicer/OrcaSlicer #4476: Can we get tree support brim width enabled on the 'default' / organic tree supports?

iandouglas, 2024-03-14:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When I choose other tree support types, I can adjust a brim setting for the tree supports, but the 'default' option (organic) does not allow this setting.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Enable the checkbox to allow auto brim, or the input field to set a brim size, per other tree support settings, and calculate the appropriate brim when slicing.

### Describe alternatives you've considered

I debated putting some primitive cylinders around where I think the tree supports are going to be for the tree supports to build upon, but that requires a "boom z distance" of 0 millimeters which might cause issues on complex models that need supports that don't start on the print bed.

iandouglas, 2024-03-14:
You can kinda work around this with "First layer expansion" but that adds expansion on everything on the first layer not just the supports.

iandouglas, 2024-06-13:
I would very much LIKE there to be activity on this. :)

MatthewMartinFD, 2024-06-13:
Bump because I would like this feature too.

sneakyjoeru, 2024-07-12:
This and density are needed

sneakyjoeru, 2024-10-12:
I believe it's still not fixed

NickiNiclas, 2024-11-15:
Just had a print fail because the tree fell. Need bigger brims for the tree.

ncarson9, 2024-11-21:
> Just had a print fail because the tree fell. Need bigger brims for the tree.

exact same issue here. Came looking for an option to simply increase the amount of brim on tree supports, but apparently it doesn't exist

Unhallowed1949, 2025-01-01:
This is one of those features where I must have scrolled through the settings 20 times looking for it, expecting it to be there but couldn’t find it because I kept missing it. No dice. This is a pretty standard feature to be customizable. Increasing the support brim width = tree supports that won’t fall over on trickier tree supports with a steep angle. Seems like a pretty necessary feature. 

Now, I’d lie to thank SoftFever for Orca. You’re awesome and I appreciate your hard work and dedication. Hopefully we can have this added :)

jagerius, 2026-10-03:
How is this still not in Orca since 2024? I want to use Organic but increase the brim so the support won't topple but I can't? Why? What is the reason for the limitation? WTF?

NickiNiclas, 2026-10-03:
@jagerius look at the number of issues, there are over 2200. This is a free project where devs donate their limited time. There are other workarounds for this but if you are eager I recommend you fork the repo and try to vibe code a solution.

Felix14-v2, 2026-10-07:
> How is this still not in Orca since 2024? I want to use Organic but increase the brim so the support won't topple but I can't?

You can use the `First layer expansion` option, that's literally the same.

[image]

But the UI inconsistency is still a valid bug/cleanup request.