OrcaSlicer/OrcaSlicer #4480: After changing the language from English to other language, any stl file cannot be sliced

dlouvard, 2024-03-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

macOS

### OS Version

Ventura 13.6.1

### Additional system information

MacbookPro M2 MAX

### Printer

Flashforge M5 pro

### How to reproduce

1. I'm trying to slice a file please with the default language, everything's fine
2. I change the language of Orcaslicer to French, and i restart the soft
3. I retry to slice the same file and now i have an alert message
[image]

### Actual results

If the language of the software is not English, the slice causes an error

### Expected results

Any language can be used for slicing

### Project file & Debug log uploads

[debug_Thu_Mar_14_12_49_10_18806.log](https://github.com/SoftFever/OrcaSlicer/files/14601659/debug_Thu_Mar_14_12_49_10_18806.log)
[bug-slicer-other-language.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14601661/bug-slicer-other-language.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

iStarComLab, 2024-03-16:
Ventura 13.6.1 - The Russian localization has exactly the same error!!!
There is no such error on version macOS 10.15.7!
[image]
[image]

Doms3n, 2024-03-18:
Same here on Mac Studio and german language. Error on slicing if the language is not english.

mrgfonkyfresh, 2024-03-24:
toujours le meme probleme avec la version 2.0.0 RC (français)

iStarComLab, 2024-03-30:
Up

dlouvard, 2024-03-30:
Always the same problem with 2.0.0 stable :-(

iStarComLab, 2024-04-04:
What is being done to solve the problem, has it taken a while to resolve this bug?

iStarComLab, 2024-04-04:
[image]

mrgfonkyfresh, 2024-04-22:
personne n a encore trouvé de solution???
apparemment sur version ultérieurs macos (catalina) cela fonctionne...

DerksenA, 2024-04-25:
[image]
Find the option Small Area Compensation, set checkbox to true and then delete all in the field Compensation Modell. After this you can disable the checkbox again and you are able to slice.

mrgfonkyfresh, 2024-04-26:
Ça ne fonctionne pas j’ai un message d erreur. 

[image]

iStarComLab, 2024-04-27:
> <img alt="image" width="402" src="https://private-user-images.githubusercontent.com/94867202/325766448-0d0955b0-2a75-4c49-8aa8-a25b8f0c0501.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQyMzAyMjMsIm5iZiI6MTcxNDIyOTkyMywicGF0aCI6Ii85NDg2NzIwMi8zMjU3NjY0NDgtMGQwOTU1YjAtMmE3NS00YzQ5LThhYTgtYTI1YjhmMGMwNTAxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI3VDE0NTg0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc1MTQwOTM2MDZmZTQ3MTc4YTU5YjcxYjM4NjIwOWJlODBlOGU3YmFlYzFjMGRhNjdiM2YzM2QxMjk1YThmODEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.JPJwZhSwKjmSRMUEQYxs5HDVR9Wqj-go9NVQ8uuGUoU"> Find the option Small Area Compensation, set checkbox to true and then delete all in the field Compensation Modell. After this you can disable the checkbox again and you are able to slice.

Everything worked out! Thank you!

iStarComLab, 2024-04-27:
[image]

iStarComLab, 2024-04-27:
[image]

closed 2024-08-03 by github-actions[bot]