OrcaSlicer/OrcaSlicer #4486: Increase size of color scheme dropdown to eliminate need to scroll

Lincoln-Jones, 2024-03-14:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The dropdown for the color scheme options is unnecessarily small and requires scrolling down to access options like Fan Speed or Temperature:

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

There is plenty of space in that area, so hopefully that dropdown element can be extended vertically so all the options are visible without the need to scroll.

kylek29, 2024-03-16:
I agree on this, the list has outgrown the initial implementation with all of the options now. 

Alternatively, I'd love to see an option to toggle on the PrusaSlicer way of using UI buttons along the bottom of the model that toggle these, it's way quicker to switch between the different views that way.

yw4z, 2025-01-25:
added to my list

closed 2025-03-30 by SoftFever