OrcaSlicer/OrcaSlicer #4488: When maximized, Orcaslicer prevents Windows Taskbar in auto-hide mode from appearing

joaomcampos, 2024-03-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0 beta

### Operating System (OS)

Windows

### OS Version

Windows 11 

### Additional system information

Windows 11 version 23H2 build 22631.3296

### Printer

N/A

### How to reproduce

1. Set windows taskbar to auto-hide.
2. Maximize Orcaslicer.
3. Go to the bottom of the screen with the mouse cursor where the taskbar should appear.

### Actual results

The windows taskbar never appears.

### Expected results

The windows taskbar should appear.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

logicbus, 2024-03-14:
I'm on Windows 11.  I do not auto-hide my taskbar.  

Several times on OrcaSlicer 1.9.x I came back to OrcaSlicer after waking my PC from sleep or using another app to find that OrcaSlicer's window reached to the bottom of the screen and covered the Windows taskbar.  On one occasion I had a different app active, and rather than see the Windows taskbar, I could see only that bottom bit of the OrcaSlicer window.  

The only way for me to resolve this was to switch to OrcaSlicer (if I hadn't already) and click the Restore Window button.  Then I could click the Maximize Window button and the OrcaSlicer button would honor the Windows Taskbar.

closed 2024-06-22 by github-actions[bot]

ChairmanNemo, 2024-10-28:
I am also experiencing this issue, please re-open it. This is a huge PITA and OrcaSlicer is the only app I use that doesn't properly deal with auto-hide. In fact I think it's the only app I've ever used that doesn't properly handle the auto-hide task bar.

aspyr, 2025-03-20:
I'm dealing with this issue as well. I noticed in OrcaSlicer.conf that when maximized the "window_mainframe" is set to: "-7; -7; 2574; 1454; 1".  This makes sense as the window seems to have a negative padding which covers the taskbar.

stipe-jurkovic, 2025-04-02:
I have teh same issue

vinniecm, 2025-04-24:
> I'm dealing with this issue as well. I noticed in OrcaSlicer.conf that when maximized the "window_mainframe" is set to: "-7; -7; 2574; 1454; 1". This makes sense as the window seems to have a negative padding which covers the taskbar.

Thanks for this breadcrumb of info! Found the file and experimented with different values, found one that made the taskbar show after un-hiding it with OS in Fullscreen mode. Mine is formatted as: "window_mainframe": "3830; 320; 1934; 1060; 1",