OrcaSlicer/OrcaSlicer #4489: Add profiles for Prusa Mk3.9?

VorpalBlade, 2024-03-14:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I notice there are profiles for the the Mk4 (with input shaping) but no profiles at all for the Mk3.9. In PrusaSlicer these have different profiles, probably due to the difference in stepper motor resolution.

### Which printers will be beneficial to this feature?

Marlin

### Describe the solution you'd like

I'm not sure how much work it is (if you can just import profiles from upstream PrusaSlicer or not). I would be willing to work (I am a C++ and Rust programmer professionally) on this assuming it isn't a lot of experimenting and tuning of profiles involved. That is, given some direction as to where and what needs to be done.

### Describe alternatives you've considered

Just continue using PrusaSlicer instead is an option. I would love to try out other slicers though.

cochcoder, 2024-03-24:
If you go to resources/profiles_template you can essentially just copy and paste the files and fill out the values. Use other profiles to guide you, it's not complicated to support a new printer, but it does require time and patience to implement one.

VorpalBlade, 2024-06-24:
I believe that Mk3.9 could use Mk4 profiles with an expanded note to the printer model check (at least this is what PrusaSlicer does):

```
PRINTER_MODEL_MK4IS
PRINTER_MODEL_MK3.9
PG
```

closed 2024-09-30 by github-actions[bot]

anghelos, 2025-03-11:
While I think I remember having found differences in the profiles in the past, it looks like the MK3.9S and MK4S profiles are now identical in PrusaSlicer, from what I can tell.