OrcaSlicer/OrcaSlicer #4491: Seam diagonal

streetgt, 2024-03-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0 Beta

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

Not revelant

### Printer

Not revelant

### How to reproduce

1 - Add an object in the build plate with the following settings, as you can see the seam is vertical.
[image]

2- Use the "clone" feature and check now the seam, the seam is on diagonal.
[image]

3- If you remove the cloned object the issue still, the only way to fix it is closing the software and opening it again.

### Actual results

Seam is on diagonal.

### Expected results

Seam must be vertical.

### Project file & Debug log uploads

Logs not needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

No

streetgt, 2024-03-22:
So I wentfurther and tested with a file .3mf and .step. The issue only happens with .step file. I have attached both. It's important to fix this issue because .step is also highly used!

[image]

[Test files.zip](https://github.com/SoftFever/OrcaSlicer/files/14727481/Test.files.zip)

qocaier, 2024-03-22:
I have the same problem. I too think that it's very big bug and it's important to solve them.

streetgt, 2024-03-22:
> I have the same problem. I too think that it's very big bug and it's important to solve them.

Yes, initially I tought it was related with Scarf seam but it isn't. Should be fixed.

closed 2024-06-30 by github-actions[bot]