OrcaSlicer/OrcaSlicer #4492: New Scarf Seam Joint produces G-Code goes beyond max print height error when scarf Start Height >0

northcoastam, 2024-03-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-beta

### Operating System (OS)

Windows

### OS Version

11

### Printer

Bambu Lab P1S

### How to reproduce

1. Load model
2. Change layer height to 0.12
3. Set Scarf Seam to "Contour"
4. Slice - no error
5. Change Scarf Start Height to 1
6. Slice - g-code beyond max height error

### Actual results

[image]

### Expected results

No error message

### Project file & Debug log uploads

[D-cap.zip](https://github.com/SoftFever/OrcaSlicer/files/14607820/D-cap.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

When searching I found similar issue #4452 but in this case Arachne is not involved.

SoftFever, 2024-03-16:
The direct cause of this error is because you have set the start height to 1 while you layer height is 0.12mm.
Make it lower than your layer height should solve the problem.
[image]
I will add a safety check later as we shouldn't set the higher than the layer height

closed 2024-03-16 by SoftFever

northcoastam, 2024-03-16:
Oops, I totally misinterpreted that setting, I thought it was kind of like "no fan for the first 3 layers" ie no scarf seams for the first 1mm of height.  Sorry for the noise and thanks for all your great work.