OrcaSlicer/OrcaSlicer #4493: Recursive Gap Fill

CCS86, 2024-03-14:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I know that many people think that Arachne is "the way forward" but I am not so convinced. I find many examples where it does not work optimally.

- It tends to inhibit arc fitting
- It can increase print time
- It can decrease quality

 I think that refining the classic perimeter generator can provide better overall results, if you don't need super thin features.

The biggest room for improvement in my eyes, would be to add "recursive gap fill" and disallow "perimeter overlap" which is currently allowed (correct me if I am wrong) 80% overlap in external perimeters and 20% in other perimeters. These overlap amounts are not flow corrected. So, in areas where there are long runs of parallel perimeters, can cause massive over-extrusion problems. 

Superslicer  added settings to control maximum allowed perimeter overlap, so that helped avoid over-extrusion. But often this just force a gap fill so wide, it could not be printed with correct fusion. The classic perimeter generator must work in complete loops (or pairs of extrusions). So, if we disallow un-compensated overlap, we inevitably find ourselves with a gap that is just shy of double the defined extrusion width:

[image]

Then, the current gap fill behavior is to fill that with a single extrusion:

[image]

I think it is relatively obvious that this will not be a successful extrusion. Even if capped by maximum volumetric rate, the kinematics of trying to flow plastic rapidly into that form factor just won't work. You will most likely under-extrude that bead, failing to bond on either side, cause excessive nozzle pressure, extruder motor heat, etc.

Instead, if we define a maximum gap fill extrusion width, and ensure that the gapfill is appropriately adjacent to the last perimeter line, we can fill every gap at, or below, this maximum gap fill width:

[image]

Then, if we run the gap filling algorithm one more time, we will get all remaining gaps:

[image]

This seems like the easiest path towards full control of extrusion width, while filling all gaps, and with no defined over-extrusion to "cheat" our way there.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

...

vgdh, 2024-03-15:
Also I often get underextrusion after gaps filled. My thought is that fill gaps pulls out more filament and when it start print next perimeter i get underextrusion . When I disable fill gaps it behaves fine.

kylek29, 2024-03-16:
The pin-holes on arcing walls; where two walls merge it leaves a non-filled gap at the intersection, I've seen it mentioned numerous times, but no fix; that you get with Arachne is the reason I switched back to Classic, so I welcome anything that wants to make it better.  I've personally always found Arachne to be better in terms of speed (sometimes as much as half the time), it's just those dang holes. It's not so much of a problem if you use right angles, but I do like my fillets / rounded corners.

CCS86, 2024-06-15:
I still think this would be very useful.

CCS86, 2026-10-03:
I still think this would be an awesome improvement. @SoftFever what do you think?