OrcaSlicer/OrcaSlicer #4494: Z-hop not working in v2 beta

smwoodward, 2024-03-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

macOS

### OS Version

14.3.1

### Printer

Ratrig 

### How to reproduce

Look for the G1 Z**** in the gcode, I was not able to find any z movement codes with z hop on both with and without firmware retraction.

### Actual results

Z hop isn't in gcode

### Expected results

z hop should be in the zip code

### Project file & Debug log uploads

[debug.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14608694/debug.3mf.zip)
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14608695/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Naahuel, 2024-03-15:
I came here looking if someone had reported this because it was driving me crazy. I couldn't make z hop work! No matter how much I tweaked the settings, nothing. 

I was collecting the evidence to add to your bug report when I realized what happened. I had changed the setting overrides for my material profile. 

I opened your debug.3mf project and what do you know...
[image]

Thank you for making me find my problem and I hope this fixed yours as well!

smwoodward, 2024-03-15:
> I came here looking if someone had reported this because it was driving me crazy. I couldn't make z hop work! No matter how much I tweaked the settings, nothing.
>
> I was collecting the evidence to [...]

🤦🏼‍♂️🤦🏼‍♂️🤦🏼‍♂️🤦🏼‍♂️🤦🏼‍♂️🤦🏼‍♂️🤦🏼‍♂️. Really......  That's what the issue was?  Now I feel dumb.

Yeah I just verified that setting was set, and also verified in the gcode by removing it, now I see the z hop.  

Well maybe we can turn this from being a bug in OrcaSlicer to being a bug in the included profiles.  lol

Naahuel, 2024-03-15:
Happens to the best of us, it was a huge coincidence you reported it first because I came here to do that. It was driving me nuts.

> Well maybe we can turn this from being a bug in OrcaSlicer to being a bug in the included profiles. lol

Yeah well I changed mine too because it came with 0.4mm as the default and I remember I couldn't see why it was z hopping! Then I forgot I fixed that on the material profile and now that I WANTED to use z hop I couldn't remember this.

SoftFever, 2024-03-16:
Awesome.
Seems the problem is solved.
Don't forget to close the issue

closed 2024-03-16 by smwoodward