OrcaSlicer/OrcaSlicer #4495: Scaling tool broken - maybe related to a painted model

tlhintoq, 2024-03-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0dev

### Operating System (OS)

Windows

### OS Version

windows10x64

### Additional system information

11th gen i9 Intel, 64gig ram

### Printer

Any

### How to reproduce

1. One model, painted top and bottom faces
2. Clone it a few times and lay out the plate
3. Want to thicken the object from 3.5mm to 5.5mm, but make no other changes.
4. With `[__] Uniform Scale` unchecked, change the Z dimension.

### Actual results

Looking at the photo we can see that in one respect Orca knows the model is 3.5mm tall: That's what it shows as the Z dimension.  YET it also places the blue Z control handle horizontally.  It has lost continuity in what direction the part is oriented.

Bug Part 2 - with uniform scale unchecked and only change the Z value, both X and Y have also changed value.  All three dimensions have changed in % when only Z should have

### Expected results

This flat coin should have changed in the correct direction as indicated in the numeric value field.
The control handles should correspond to the numberic value fields.
X & Y should not be changing when uniform scale is unchecked.

### Project file & Debug log uploads

This part is IP and can't be released.  If you are unable to reproduce let me know and I'll make a generic part version to submit.

### Checklist of files to include

- [X] Log file
- [X] Project file

tlhintoq, 2024-03-14:
[image]

vgdh, 2024-03-15:
confirm, its broken

tlhintoq, 2024-04-15:
Version 2 stable release.
I can't make it fail even with a painted object.  But this ticket hasn't been noted as fixed or merged.  So maybe a fix in something related also fixed this for free.  Something to keep an eye on.

closed 2024-07-22 by github-actions[bot]