OrcaSlicer/OrcaSlicer #4497: Option to Completely Disable the Device Tab

charredchar, 2024-03-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

With the device tab being loaded at all times it puts unnecessary strain on the printer host.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

An option in the Physical Printer Connection settings to completely disable the Device tab. Not only to prevent it from loading the printer host webpage at all but to also hide the tab.
Preferable the setting would be, "Enable Device Tab" that needs to be opted into and not something that requires the user to opt out of after the fact.

### Describe alternatives you've considered

A secondary setting for;
"Do not load Device Tab in the background" would also be a bonus. So that some can still have use of the tab but don't have it load the printer host page until the user clicks on it. Also have it unload when leaving the device tab, maybe a timeout to account for fast tab switching.

### Additional context

I would prefer having the option to remove the tab entirely along with preventing it from loading anything ***much more*** than anything else. I don't intend to and never will use the device tab and having another place that is constantly putting load on my Klipper host is not great. Not to mention if, like me, you are in the habit of leaving the slicer open it puts more load on the PC, OrcaSlicer constantly uses more than twice the memory as other slicers due to this tab being loaded.

closed 2024-06-22 by github-actions[bot]

jrsall92, 2024-10-05:
I've noticed that the device tab when load is makes my laptop lag significantly (MBP M1 Pro/16GB). I'd also like to see the oprional removal of the tab as well.

charredchar, 2025-01-11:
I asked awhile ago about this on Discord and was basically told it'll never happen. So if for any reason you don't like the fact the Device Tab is always loaded in the background you're pretty much SOL it seems.

scratchboom, 2025-07-09:
As a workaround just use "about:blank" as "Device UI" in printer connection settings

[image]