OrcaSlicer/OrcaSlicer #4498: Mirroring STL causes crash

DBShrill, 2024-03-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1 and 2.0.0 beta

### Operating System (OS)

macOS

### OS Version

MacOS 136.2

### Additional system information

M2 Pro 16gb

### Printer

Voron 0.2

### How to reproduce

1. Import stand_2_b_LHS.stl
2. Lay flat using Auto orient
3. Mirror across x-axis
4. Try to slice

### Actual results

App crashes
[image]

### Expected results

Should be able to slice normally and upload it to my designated printer

### Project file & Debug log uploads

[stand_2_b_LHS.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14610752/stand_2_b_LHS.3mf.zip)
[debug_Fri_Mar_15_11_30_25_61421.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/14610755/debug_Fri_Mar_15_11_30_25_61421.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SoftFever, 2024-03-17:
I'm not able to repro the bug with the attached project.
Are you able to repro this crash consistently? Any extra info would be helpful

Sushiman42, 2024-04-03:
I am having trouble with this as well. Seems to be related to #3346. Interestingly, I was able to solve this by mirroring my parts, exporting as STL, and then re-importing them alongside the non-mirrored originals.

DBShrill, 2024-04-03:
> I'm not able to repro the bug with the attached project. Are you able to repro this crash consistently? Any extra info would be helpful

Only for that specific file. I provided all the information about that specific error but a problem that arose was that if my layer height was above 0.32 (for a 0.4mm nozzle), the app would never open. I could only fix this issue by manually editing the slicer profile using some text editor. 

As of now I have not had any error with any other file the 2.0.0 beta seems to be working well.

DBShrill, 2024-04-03:
I have tried reproducing the error again and have not been able to. If someone else were to have this issue I would suggest trying to manually lay the object flat or mirroring the object directly from a different modeling software. As far as I know, this seems to be an isolated issue that may have only affected me.

closed 2024-04-03 by DBShrill

BlackEagle94, 2024-04-07:
The issue still Exists on v2.0.0 nightly build.
The Object was Mirrored correctly. When Start to Slice Orca will Crash.
[Orca v2.0.0 Mirroring Crash.txt](https://github.com/SoftFever/OrcaSlicer/files/14897946/Orca.v2.0.0.Mirroring.Crash.txt)

ckonkel, 2024-05-18:
I have the same issue with this model, when I try mirroring (x or y or z)it I get this error:
Orca 2.0.0 windows 11

A fatal error occurred: "Access violation"
Please save your project and restart the application

using this model

[tb2-side-front-v2 (1).zip](https://github.com/SoftFever/OrcaSlicer/files/15358984/tb2-side-front-v2.1.zip)

have seen other errors that force me to restart orca as well, but have not replicated , but happens when mirroring