OrcaSlicer/OrcaSlicer #4502: Print infill first not working properly on mixed objects

Jucaer85, 2024-03-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1 (also 2.0.0 beta)

### Operating System (OS)

Windows

### OS Version

11 Pro

### Printer

Bambulab X1C

### How to reproduce

1. Select at least 3 object
2. Set "Print infill first" only on 2 of them.
3. Check the print path on a random layer

### Actual results

By checking the print path, you'll notice that - out of the 2 selected objects to be printed with infill first - one will be printed infill first, the other one walls first.

### Expected results

Out of all the 3 imported objects, the 2 selected ones should be printed with infill first before the walls.

### Project file & Debug log uploads

[log files.zip](https://github.com/SoftFever/OrcaSlicer/files/14616484/log.files.zip)
[project files.zip](https://github.com/SoftFever/OrcaSlicer/files/14616485/project.files.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

timX24968B, 2024-04-06:
I would like to add to this issue another related problem:  when there is a single object with multiple isolated sections, it will still print "infill - wall - infill - wall" instead of "infill - infill - wall - wall" when "print infill before wall" is selected.  this disrupts the calculations the software makes when calculating waste, which results in the color bleeding through when enabling "flush into object's infill" because it calculated printing through the entire part's infill when the program ends up printing walls before other parts of the infill during slicing.  This results in some of the walls being made from partially flushed material, and thus, discoloration.

crowding, 2024-06-16:
+1. I encountered this when I was testing the effect of "perimeter first" on bridging. Attached .3mf file, the one marked "infill first" slices with the perimeter first and vice versa.

[infill first bug.zip](https://github.com/user-attachments/files/15855815/infill.first.bug.zip)

closed 2024-09-22 by github-actions[bot]