OrcaSlicer/OrcaSlicer #4505: Printer selection not working, printer/preset does not show up after selecting.

Boerny41, 2024-03-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0

### Operating System (OS)

Windows

### OS Version

Win 10

### Printer

Anycubic Mega SE

### How to reproduce

1. try to add new Printer
2.  it does not show up

### Actual results

github only allows 10MB. https://imgur.com/a/GdIraqm

### Expected results

the printer showing up

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14617377/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Boerny41, 2024-03-16:
Yes, I did, and the printer did reappear after a restart, but I later noticed that it also disappeared in the middle of a session, for example when I added a custom filament to the printer. Again, it reappeared after a restart, but since this does not happen when adding a filament to a Klipper printer preset, but only to Anycubic printer presets, I decided to leave this open.

spatiela, 2024-03-17:
Same situation with the FlashForge Adventurer 5M. It shows under printer options but not under physical printer but it was on that list on other people's computers. Is it because I am on a Mac? Sorry, I am really new at this and most of the coding language is totally foreign to me 
[image]
[image]

closed 2024-06-24 by github-actions[bot]