OrcaSlicer/OrcaSlicer #4510: SEMM-Output variable volume purge info with "purge into prime tower" disabled

53Aries, 2024-03-16:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I am attempting to purge into a bucket using variable volume purge. Orca will not output variable purge volumes to gcode unless "purge into prime tower" is enabled. With "purge into prime tower" enabled I am unable to purge into bucket AND have a prime tower. All purging is done into prime tower. 

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

I would like to do like a Bambu printer and purge into bucket using variable volume purge and also use a prime tower after purging. I see in the Bambu X1 preset there is an option in print settings under prime tower for "prime volume" which is not present in generic klipper print settings. Also for the Bambu preset, Orca is outputting variable volume purge info while "purge into prime tower" is disabled. 

### Describe alternatives you've considered

I have found no way to use variable volume purge without doing so into a prime tower. Currently I do not use a prime tower so that my purge can be done in a bucket.

53Aries, 2024-03-16:
This is the same issue as https://github.com/SoftFever/OrcaSlicer/issues/2900. It seems it was fixed for Bambu printers but not other printer profiles.

Batalhoti, 2024-04-18:
I had the same problem as you, the alternative was to go back to version 1.8beta, where you can use the same filament change gcode as bambulab to purge in the bucket.

closed 2024-07-25 by github-actions[bot]