OrcaSlicer/OrcaSlicer #4511: crash during loading project

Rhoenbussard, 2024-03-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

 OrcaSlicer_Windows_V2.0.0-beta_portable - latest nightly build

### Operating System (OS)

Windows

### OS Version

win 10

### Additional system information

Amd Radeon R7

### Printer

Sovol SV01

### How to reproduce

load a project
import a file

### Actual results

Loading anyy project
The loading of the project starts. About 60% the programm crashed.
Same happens during importing a new file

### Expected results

show the project

### Project file & Debug log uploads

[Shape-Box.zip](https://github.com/SoftFever/OrcaSlicer/files/14623059/Shape-Box.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

vgdh, 2024-03-16:
crash on this [project](https://filetransfer.io/data-package/5yjD7Mu6#link)

SoftFever, 2024-03-17:
Hi 
Can you give the latest nightly build again?
I've fixed a crashing bug this morning. It should work now

vgdh, 2024-03-17:
> Hi Can you give the latest nightly build again? I've fixed a crashing bug this morning. It should work now
works with my project

closed 2024-06-24 by github-actions[bot]